Ultimaker/Cura #18808: Slice failed when using support blocker

Yankeeki2024, 2024-04-03:
### Cura Version

5.6 or 5.7

### Operating System

Windows 11

### Printer

Ulitimaker

### Describe model location

any project

### Describe your model

dino

### Add your .zip here ⬇️

why cant I attach a project file?  doesn't accept .3mf?

Yankeeki2024, 2024-04-03:
[UMS5_Main_Dino.zip](https://github.com/Ultimaker/Cura/files/14857144/UMS5_Main_Dino.zip)

HellAholic, 2024-04-03:
[UMS5_Main_Dino.zip](https://github.com/Ultimaker/Cura/files/14857434/UMS5_Main_Dino.zip)
Can you check and see if you are able to slice this one?

Yankeeki2024, 2024-04-03:
yes.  But as soon as I try to add a support block I get an error.

[image]

Yankeeki2024, 2024-04-03:
I also followed instructions above and deleted and all instances and reinstalled version 5.3.  Issue did not resolve.

HellAholic, 2024-04-03:
Support blocker is a mesh and by default gets assigned to the first extruder.
You are disabling the first extruder on your printer and adding support blockers.
Although a support blocker is not printed it is a mesh, the initial check for whether any type of mesh is assigned to a disabled extruder fails.
You can either switch the extruder of the Support blocker meshes to extruder 2, or enable extruder 1 and have not touch support blockers.
In either case it will slice.

Yankeeki2024, 2024-04-03:
You are absolutely correct.  I've been using nozzle 2 to use up some old PLA.

Thanks

ro-mak, 2024-05-08:
> Support blocker is a mesh and by default gets assigned to the first extruder. You are disabling the first extruder on your printer and adding support blockers. Although a support blocker is not printed, it is a mesh; the initial check for whether any type of mesh is assigned to a disabled extruder fails. You can either switch the extruder of the Support blocker meshes to extruder 2, or enable extruder 1 and not touch support blockers. In either case it will slice.

thanks, changing the extruder for the eraser helped
I think it will be great to either automatically set the available extruder or at least write a warning message describing the reason why Cura is not able to slice