Ultimaker/Cura #18809: SV04 Single Mode 2 Not Working

drwho83, 2024-04-03:
### Cura Version

5.6.0

### Operating System

Windows 10

### Printer

SV04 - Tried with Both Stock and 3rd Party Firmware

### Reproduction steps

I was printing fine with the previous version of Cura. I updated to the latest version. Sliced a new file and saved it to the SD card. I then put the card in the printer. Instead of heating up the 2nd print head (this is a IDEX printer) it started heating up the 1st print head. I double checked I had the right one selected. I have single mode 2 selected just like I always do when I want to only use the right (2nd) print head. I also hadn't changed the mode on the printer. It's on single/nozzle 2 (right extruder).

This is very strange. I'll just use prusa slicer for now and wait until the next version of cura comes out unless there's a patch or something I can install.

### Actual results

The first (left) extruder heats up instead of the 2nd (right) extruder. 

### Expected results

I expect the right extruder to heat up and the left extruder to stay parked on the left side while the right extruder prints.

### Add your .zip and screenshots here ⬇️

[Model.zip](https://github.com/Ultimaker/Cura/files/14859041/Model.zip)

GregValiant, 2024-04-04:
Thanks for the report.
When I loaded a SV04 Single Mode 2 printer it only has a single extruder and it is extruder 1.  That means the gcode is correct as nothing seems to be telling Cura that you have a dual head printer.

I'll remove the bug label and close this.  I have seen problems with the other Sovol SV04 printer definitions as well.  The definitions come from the manufacturer or from an owner who submitted a definition file.  I hope you understand.

What you could try doing with the "single mode 2" printer is to put "T1" as the first line in your startup gcode.

closed 2024-04-04 by GregValiant

drwho83, 2024-04-05:
Thanks and I think I do understand.

That's valuable information and hopefully if it gets updated in a future
version of cura The source will provide the correct info.

The previous version doesn't but the latest and greatest recently released
version of Prusa slicer has some random lines of junk in the default G-Code
for some reason for the sv04.

I cleaned it up and let them know but it didn't seem to do anything at all.
Why it was there though I have no idea.

Again, thanks! Appreciate the reply!

GregValiant, 2024-04-05:
I looked into this some more.  At the base of the problem is this fact:
**"Cura does not support IDEX printers."**
Which brings up the question:
**"How do you slice the models for IDEX printers using Sovol Cura, or UM Cura?"**

The Sovol solution seems to be to use multiple printer definitions.  The Raise3D solution is to use a special Gcode command "M605 S0, S1, or S2".

For Sovol, each printer definition changes some basics in Cura.  If you load the "Mirror" or "Copy" machine you see that the build plate is 1/2 the normal width and provides a single slice of the model.  If you open the gcode, there is a single model.  No mirror image or second print is present in the slice.  Something must tell the printer which mode to use.
The only difference I see in a Sovol gcode is this line:
";TARGET_MACHINE.NAME:SV04 Copy Mode" or
";TARGET_MACHINE.NAME:SV04 Mirror Mode" or
";TARGET_MACHINE.NAME:SV04 Dual Mode" or 
";TARGET_MACHINE.NAME:SV04 Single Mode 01" or
";TARGET_MACHINE.NAME:SV04 Single Mode 02" or

So it appears that when the machine sees that line, it sets the mode within the printer mainboard and then the printer performs the translation of the gcode into the proper form and translates the X and Y axes movements into U and V axes movements.

The difference between Sovol Cura and UM Cura is where those lines go into the Gcode.
For Sovol Cura it's line 2 like this:
;FLAVOR:Marlin
;TARGET_MACHINE.NAME:SV04 Single Mode 02
;TIME:3768

While in UM Cura the line goes in down further at line 11:
;MAXZ:24.9
;TARGET_MACHINE.NAME:SV04 Single Mode 02
;Generated with Cura_SteamEngine 5.7.0

That leads us to this.  You are going to be the guinea pig.
This is a post processor that cuts the "TARGET_MACHINE.NAME:" line from line 11 and pastes it into line 2.
Unzip the file and put the SV04_IDEX.py file into your configuration folder in the "scripts" folder.  It will be available with the other post processors.  Each of the "Sovol SV04" printers you have installed will need this active when you save the gcode.
I don't own one of those printers so I can't tell if it will affect how the printer starts.  Let me know if it works.

[Sovol_SV04_IDEX.zip](https://github.com/Ultimaker/Cura/files/14884847/Sovol_SV04_IDEX.zip)