Ultimaker/Cura #18811: Nothing is slicing on 5.7

itsnotcake, 2024-04-03:
### Cura Version

5.7

### Operating System

windows 11

### Printer

ender 3s1 pro

### Describe your model

coaster no holes i have tried all modes all show same error

### Add your .zip here ⬇️

[image]

wawanbreton, 2024-04-04:
Hi @itsnotcake, thanks for the report !
This issue is very annoying, there must be something in your environment/OS that prevent the slicing engine to run properly.
Did you run previous versions of Cura before ?
Anyway, I can only give you some global advice to try and fix this: 
* Restart Cura
* Restart your PC
* Reinstall Cura

If nothing helps, please [send the log files](https://support.makerbot.com/s/article/1667337559413) so that we can further assist you.

DevelopmentAnything, 2024-04-04:
Reinstallation does not help.
New installation of 5.7 also causes the same problem.
To resolve this, you can slice it by replacing it with another profile and then loading the original profile again.
Without changing the profile's detail settings, changing the profile will enable slicing, which is clearly an internal issue for the cura.

Vandresc, 2024-04-05:
Maybe it is something in your configuration? Did you adjust any settings? And is it possible to share the project file which hopefully will help us reproduce the issue?

itsnotcake, 2024-04-05:
Thank you everyone for your help! The solution was adding the printer again and deleting the old one I don’t understand why that worked but it did.

Vandresc, 2024-04-05:
Maybe something in your printer definition that was adjusted? Anyways, good that you resolved it and thank you for replying.

closed 2024-04-05 by Vandresc

wawanbreton, 2024-04-05:
I'm glad that you could solve your issue !
It shouldn't have happened though, so we'll keep an eye on it if similar reports pop up, but that may be an isolated incident.

itsnotcake, 2024-04-05:
> I'm glad that you could solve your issue !
> It shouldn't have happened though, so we'll keep an eye on it if similar reports pop up, but that may be an isolated incident. 

For more context to help I had a previous version of cura 5.3.1 so maybe when it was transferring the information to 5.7 it was corrupted?

wawanbreton, 2024-04-05:
Thanks for the precision, that gives more useful context indeed.
Technically we have upgrade scripts for every version change, and they are supposed to be applied one by one when upgrading from a very old version, so that should theoretically not happen.
But in practice, there can be "bugs" in some upgrade scripts, so if you have many of them to apply, a corruption is more likely to happen.
When possible, the safest option is to clear your configuration completely when something goes wrong, but that may be very annoying if you have custom profiles/materials.