Ultimaker/Cura #18813: Manage Printers failing

mikimouse13579, 2024-04-04:
### Cura Version

5.7.0

### Operating System

MacOS Big Sur

### Printer

Sovol SV06

### Reproduction steps

1. Updated application
2. It was announced that Sovol printer is upgraded
3. Pressed Manage Printers the screen appears correctly
4. Selected "Machine Settings"
5. Close "Machine Settings"
6. the only way to get "Manage Printers" screen again is to close Cura application

### Actual results

if you follow above you are not able to call "Manage Printers" more than once, application has to be closed and reopened again to be able to see "Manage Printers" screen again

### Expected results

I would expect that every time I would press "Manage Printers" I will see screen where I can do something

### Add your .zip and screenshots here ⬇️

no need for the screenshot as I am not having error or wrong screen I am missing screen

wawanbreton, 2024-04-04:
Hi @mikimouse13579, thanks for the report !
I'm afraid this is a real bug but specific to MacOS. It reminds me of something very similar we fixed recently.
I'll report it internally and see if you can fix it. In the meantime I'm afraid there is no workaround to this. Fortunately, I hope you don't open the "Manage Printers" screen everyday.

wawanbreton, 2024-04-04:
Well apparently the issue is already known internally, and I was told that the "Manage Printers" window actually appears, but behind the Cura main window. So you have to manually switch to it to make it visible, but at least that should be possible without having to restart Cura.
Can you confirm that you have the same behavior ?

mikimouse13579, 2024-04-04:
hi

thank you very much for prompt reply,
I do not open "Manage Printer" often I was curios if I will see changes as
I am using SV06 every day :)

Just wanted you to know,

thank you
Marek

mikimouse13579, 2024-04-04:
yes I can confirm that the form is created in the background and
minimising application I can see it and use it

thank you so much :) :) :) ;)

wawanbreton, 2024-04-04:
> I am using SV06 every day :)

I can understand that very well :smiley: 

Very nice that you could retest, that confirms that this is the same issue. Not that annoying then, but at least it's flagged and we will fix it at some point.

Thanks again for reporting !

closed 2024-04-04 by wawanbreton