Ultimaker/Cura #18814: 5.7.0 gcode hangs Sidewinder X1, where v5.6 gcode works fine.

frozentimeimages, 2024-04-04:
### Cura Version

5.7.0

### Operating System

MacOS

### Printer

Artillery Sidewinder X1, stock

### Reproduction steps

Sliced simple STL file.
Tried to run on Sidewinder X1 printer.
(I then sliced with v5.6, and everything went fine. This was reproducible with other models: v5.6 OK, v5.7 hangs)

### Actual results

The Sidewinder hung, the clock on the display kept running, but no attempt to print was made. Response to buttons was also sluggish. Felt like processor was in a tight loop.
This state continued for 20 minutes when I turned off the machine. Same result with different STLs.
Printer still runs fine with v5.6 files, new and old.

### Expected results

The printer should have started printing...
Buttons should have responded quicker.

### Add your .zip and screenshots here ⬇️

[ASX1_Waste pipe collar small.3mf.zip](https://github.com/Ultimaker/Cura/files/14863785/ASX1_Waste.pipe.collar.small.3mf.zip)

GregValiant, 2024-04-04:
Thanks for the report.
I generated a gcode of the project and it looked fine.  The gcode shows the nozzle and bed heating and the print starts.

Post an "old gcode" and a "new gcode".  Maybe they will show a difference.  (You will need to zip them.)

GregValiant, 2024-04-04:
Yes it does help
For some reason, the Artillery printers rely on post processing to be able to print.  That is really dumb.  If you put "M73 P0" into your startup gcode it may work.  The "print percentage" on the LCD won't change, but it should print.

This bug goes on me because I obsoleted "Display Progress On LCD" and "Display Filename on LCD" and included them in "Display Info On LCD" which is new for 5.7.0. 

Looking at your gcode, you can see the M73 line goes in as the second line in the gcode.  That is actually not allowed because other post processors will look for the Time line and it has been pushed down.  Printers like Creality don't like that.

If you load the post processor "Display Info On LCD" and have it set to "Display Progress" you will see that the M73 line is available to be inserted.  With the "P" parameter it starts the percentage counting up...maybe.
M75 and M77 might also be involved.  There is no way to tell what the Artillery people were thinking when they did this.

If you are using Octo-Print you might also have to add the M118 line with the A and P parameters.

This problem was brought up previously in bug report #18766 and I have made the changes to the file but not in time for 5.7.
I'll mark this as a duplicate.

It is really odd that the printer won't run unless the Print Time Clock is turned on and neither your definition file nor your printer do it as a matter of course.
If you continue to have a problem let me know.  I don't have one of those printers and debugging from afar isn't easy, but I can give it a shot.

frozentimeimages, 2024-04-05:
Hi,
And thanks again for your rapid and thorough response. It's amazing that you guys manage to keep this code supporting so many, and such old printers.
It turns out that all I needed to do was activate the new DisplayProgress extension, with M73 checked.
Running the 5.7 beta with that enabled, everything works fine, including showing progress, and shutting down.
So case closed from my perspective.
Thanks again, and keep up the good work.
Mark

closed 2024-04-05 by Vandresc