Ultimaker/Cura #18818: Help----5.7.0 issue with linux

oldfart1970, 2024-04-04:
### Cura Version

5.7.0

### Operating System

linux on chromebook

### Printer

anycubic

### Reproduction steps

Previous version worked great, updated to 5.7.0 yesterday now getting The Wayland connection broke. Did the Wayland compositor die?

### Actual results

nothing ran several times 

### Expected results

.

### Add your .zip and screenshots here ⬇️

⬇️.

riegojerey, 2024-04-07:
well you can try to boot to x11, if it is okay then maybe wayland broke

nallath, 2024-04-15:
Please include the log files.

kafufafu, 2024-04-25:
If you need help running the AppImage, I can provide you with the instructions that worked on my system (ChromeOS Flex running Linux from the Penguin terminal)  

:)

oldfart1970, 2024-04-26:
Yes please that would be very appreciated. i was able to load bambuu labs slicer will that cause a problem to have them both in linux?

thanks,
Doug

kafufafu, 2024-04-26:
Alright, I'll give you the instructions that have worked for me:

1) First, execute the following commands in the ChromeOS Flex terminal (within Penguin):
`sudo apt install fuse`
`sudo apt install flatpak`
and
`flatpak --user remote-add --if-not-exists flathub https://dl.flathub.org/repo/flathub.flatpakrepo`

2) Restart Penguin (or close the terminal and restart ChromeOS Flex if you can't find the button).
3) Open the terminal, run Penguin, and install the following from Flathub:

`flatpak install flathub it.mijorus.gearlever
`

4) Download the official Ultimaker Cura AppImage (5.6.0 as of April 26, 2024):

[https://github.com/Ultimaker/Cura/releases/download/5.6.0/UltiMaker-Cura-5.6.0-linux-X64.AppImage  ](url)

5) Move the downloaded file to the "Linux files" folder.

6) Run GearLever and go to "home"; you should see the Cura AppImage you moved to the root of the Linux folder.
Press "unlock" and then "open." If Ultimaker Cura opens without issues, you can install it to create a shortcut in the ChromeOS apps menu.
(GearLever was installed in step 3)

Hope this helps!"

Greetings from Chile ;)

MOBJ2R, 2024-06-02:
Stumbled upon this post as I am running into the same issue. 

I followed all the steps above. When I try to launch Cura in gear lever I hit "unlock" and then nothing else pops up and I cannot do anything. Any advice on how to open it? Thanks! @kafufafu

kafufafu, 2024-06-02:
@MOBJ2R 

1) The issue you're describing seems to stem from the absence of FUSE. Due to significant issues I encountered with it, I recommend reading the updated comment with new information.

2) There's definitely an issue with Ultimaker Cura's code post version 4.13. Currently, the problem persists on many Linux distros and some systems running Windows 11.

_3) I decided to leave ChromeOS Flex and try Ubuntu 24.04 LTS. I've had a very pleasant experience with the system, so much so that I personally don't consider going back to Windows or ChromeOS._

3.1) The only way to have Ultimaker Cura running properly and without issues was by installing version 4.13.2 from the Ubuntu APPS store.

3.2) I hope you can find the right path and resolve it. Sadly, I've given up.

wernight, 2024-07-08:
Cura 5.6 works on Crostini (Debian-base Linux on Chromebooks), however 5.7 (including 5.7.2) doesn't.

The error has been reported in multiple places:

  - https://www.reddit.com/r/linuxquestions/comments/1c1nwg7/attempt_to_run_cura_wayland_dead/
  - https://www.reddit.com/r/chromeos/comments/1c8uhai/hi_trying_to_run_ultimaker_cura_in_linux_and_i/

Workarounds:

  - Downgrade to 5.6
  - Force X11: `QT_QPA_PLATFORM=xcb ./Cura.AppImage` (however some UI like the Marketplace seem broken).

The Cura log has a few warnings:

```
WARNING - [MainThread] cura.OAuth2.KeyringAttribute.__set__ [78]: Exception: No keyring backend present
WARNING - [MainThread] cura.OAuth2.KeyringAttribute.__set__ [78]: keyring.errors.NoKeyringError: No recommended backend was available. Install a recommended 3rd party backend package; or, install the keyrings.alt package if you want to use the non-recommended backends. See https://pypi.org/project/keyring for details.
WARNING - [MainThread] UM.OutputDevice.OutputDeviceManager.removeOutputDevice [235]: Could not find output device with id digital_factory to remove
WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [448]: file:///tmp/.mount_Cura.AyheaVb/share/cura/resources/qml/Settings/SettingView.qml:204:16: QML SettingDefinitionsModel: Binding loop detected for property "expanded"
WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [448]: file:///tmp/.mount_Cura.AyheaVb/share/cura/resources/qml/PrintSetupSelector/Recommended/RecommendedSupportSelector.qml:60:17: Unable to assign [undefined] to int
WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [448]: file:///tmp/.mount_Cura.AyheaVb/share/cura/resources/qml/PrintSetupSelector/Recommended/RecommendedSupportSelector.qml:95:17: Unable to assign [undefined] to int
WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [448]: <Unknown File>: 
```