Ultimaker/Cura #18820: Slicing error.

Fiks-007, 2024-04-04:
### Cura Version

5.7.0

### Operating System

Windows 10

### Printer

Two Trees Sapphire Pro

### Name abnormal settings

No

### Describe model location

No

### Describe your model

In version 5.6.0 it cuts without errors.

### Add your .zip here ⬇️

[Semoir_Graines_de_1__5_mm_-_Partie_Basse_-_SG40 Generic PETG lw 0.4mm lh 0.2mm if 10% ext1 240C bed 70C.zip](https://github.com/Ultimaker/Cura/files/14873313/Semoir_Graines_de_1__5_mm_-_Partie_Basse_-_SG40.Generic.PETG.lw.0.4mm.lh.0.2mm.if.10.ext1.240C.bed.70C.zip)

GregValiant, 2024-04-04:
Thanks for the report.
I didn't have any trouble slicing this on Windows10 and Cura 5.7.0.

Use the "Help | Show Configuration Folder" command.  In the folder will be the "Cura.log" file.  Zip that and post it here.  I may point to the reason for the problem on your system.

Fiks-007, 2024-04-07:
Log file.
[cura_log.zip](https://github.com/Ultimaker/Cura/files/14898044/cura_log.zip)

GregValiant, 2024-04-07:
Thanks for that.
@HellAholic I hate to keep calling you out on these.  Here is another one:
"Backend exited abnormally with return code 3221225501!"

Fiks-007, 2024-04-07:
What is the reason for this and how to fix it?

HellAholic, 2024-04-07:
[TWOTSPRO_Semoir_Graines_de_1__5_mm_-_Partie_Basse_-_SG40 Generic PETG lw 0.4mm lh 0.2mm if 10% ext1 240C bed 70C.zip](https://github.com/Ultimaker/Cura/files/14898952/TWOTSPRO_Semoir_Graines_de_1__5_mm_-_Partie_Basse_-_SG40.Generic.PETG.lw.0.4mm.lh.0.2mm.if.10.ext1.240C.bed.70C.zip)

I can slice without issue.
[image]

Just gathering some information to see if I can find a common factor here:
- Do you have any plugins installed?
- Side note, what's your system's CPU?

HellAholic, 2024-04-08:
Hey @Fiks-007 , another user made a discovery last night attempting to debug the issue.
Could you check and see if you're still getting the slice fail if you run Cura as Administrator? [Right-Click Run As Administrator]

@GregValiant FYI, this is an interesting find.

Fiks-007, 2024-04-08:
I found the cause of the error. If the file name or path to the file does not have a Latin font (for example, Cyrillic), then this error occurs. In this case, the model is cut and if you manage to click on the “save G-code” button before the error message appears, the file will be saved in the specified location.

Fiks-007, 2024-04-08:
In administrator mode the error is not resolved.

GregValiant, 2024-04-08:
A font issue?
I feel so dumb.  I can't believe it wasn't the first thing I thought of.

Fiks-007, 2024-04-08:
Most likely not with the font, but with the localization of the operating system..

Fiks-007, 2024-04-09:
We are waiting for the error to be resolved.

closed 2024-04-09 by Fiks-007