Ultimaker/Cura #18821: Slicing failed after syncing with account

SamAntUA, 2024-04-04:
### Cura Version

5.7.0

### Operating System

Ubuntu Linux

### Printer

Anycubic Kobra Go

### Describe your model

small container

### Add your .zip here ⬇️

I encountered this before too. After login to account a popup arised offering to sync with account. After syncing slicings begin to fail. Reverting cura settings using backup helps.

Vandresc, 2024-04-05:
Hi @SamAntUA , thank you for the report. Hmm, I do not get the error when following the above steps. Is it possible to share your cura user config with us or at least the project file containing the printer definition, profile with user changes that is active when the error occurs, and the model. Thank you

SamAntUA, 2024-04-19:
Hi! Sure. In Linux config you need is ~/.local/share/cura/5.7/cura.cfg ?