Ultimaker/Cura #18825: slicing fails after reloading model

RenchesAndSords, 2024-04-05:
### Cura Version

5.7.0

### Operating System

Windows 11

### Printer

tronxy x5sa

### Describe model location

center of build plate

### Describe your model

katana shaped shifter knob

### Add your .zip here ⬇️

[Katana.zip](https://github.com/Ultimaker/Cura/files/14887261/Katana.zip)

GregValiant, 2024-04-05:
Thanks for the report.
I can't duplicate this.  It slices fine whether it's the first time or after a reload.
Use the "File | Save Project" command to create a special 3mf project file.
Use the "Help | Show Configuration Folder" command to find the Cura.log file.
Zip those up and post the zip folder here.

RenchesAndSords, 2024-04-05:
I'm also having trouble recreating it as well, I guess it was just a one time thing

closed 2024-04-05 by RenchesAndSords