Ultimaker/Cura #18832: Thingiverse 'Open in Cura' not working with 5.7

gh13, 2024-04-06:
### Cura Version

5.7

### Operating System

Windows 10

### Printer

Ender 3 S1 PRO

### Reproduction steps

Cura 5.7.0 is installed on the PC.
Opened Thingiverse in Chrome, selected a model then the files tab.
Clicked the 'Open in Cura' icon. for the STL file

### Actual results

Nothing happens

### Expected results

Model should open Cura 5.7 & load the model

### Add your .zip and screenshots here ⬇️

[image]

HellAholic, 2024-04-06:
Hey @gh13 ,
The **Open in Cura** works through the Windows Registry association with links from Thingiverse. If you uninstalled other Cura versions (Beta 5.7 as an example) after installing the 5.7 you might have remove those registries. If that is the case, re-installing the 5.7 should fix the issue.
If that is not the case, check if you see some sort of pop up like the image below when you click on Open in Cura button,
[image]
This pop up might look different depending on your browser but the message is the same. Check if you blocked this somehow or did not allow for it.

gh13, 2024-04-06:
I had uninstalled the 5.7 beta following installing the release version, which may have done as you said. 
I've ran the 5.7 installer again & the Thingiverse integration is now working. Thanks :-)

closed 2024-04-06 by gh13