Ultimaker/Cura #18834: Drastically different print speeds than 5.6

kevinjones4642, 2024-04-06:
### Cura Version

5.7.0

### Operating System

Windows 10

### Printer

Ultimaker S5

### Reproduction steps

I have a part that I have printed when sliced with Cura 5.6, and now with 5.7, and 5.7 doesn't seem to honor print speed requests, or has very different material profiles. A job sliced in 5.7 lists 1 hour 33 minutes. If I save the project out of 5.7, and load it into 5.6, when I re-slice it claims 1 hour 50 minutes.

Normally speed would be good, but in this case it is PETG through a 0.25mm nozzle, and when i print fast, wall-to-wall adhesion frequently fails. A slower speed often remedies this. 

### Actual results

The head was lightening fast on the brim, so much so that the individual lines on the brim appear disconeected. 

### Expected results

I didn't expect speed to change drastically and violate commanded speeds changing from 5.6 to 5.7

### Add your .zip and screenshots here ⬇️

[UMS5_Motor Mount V4.4.3mf.zip](https://github.com/Ultimaker/Cura/files/14895132/UMS5_Motor.Mount.V4.4.3mf.zip)

GregValiant, 2024-04-07:
Thanks for the report.
I sliced the file and using MS Excel and looking at the speeds on the Initial Layer:
You have the Number of Slower Layers = 1 and a Print Speed of 40.
The entire skirt is at 30mm/sec
Wall-Outer is at 8mm/sec
The Wall-Inner varies from a low of 13mm/sec to a high of 30mm/sec.  The line width of inner walls varies the most so that makes sense to me.
Skins are at 30mm/sec.

Those speeds look to be in line with your settings.  Post that 5.7 gcode file and I can read it into Excel.

Bzzz, 2024-04-07:
Check if it's actually generating similar code - 5.7.0 here does not honour top layer count and still prints 3 layers when I ask for 5 (which are even displayed in the preview tab). Previous profile settings from Cura 5.6 were 3L x 0.33mm, current settings would be 5L x 0.20mm.
Going back to 5.6 as that completely ruins surface finish with those thinner layers.

kevinjones4642, 2024-04-08:
Quick update I think it's a profile problem. In 5.6, the default print speed for PETG with an 0.25mm nozzle is 55mm/s, but in 5.7 it is 100 mm/s. I guess I haven't paid that much attention before, I thought that changing 55 to 40 was some kind of percentage rather than mm/s. I'm now also confused that entering 40 mm/s in Cura 5.6 results in a max speed of 30 mm/s. Can anybody shed light on the true meaning of the print speed setting.

GregValiant, 2024-04-08:
- In the Speed Settings:
  - The units for all speeds are in mm/sec
    - If you have "Number of Slower Layers" set to 1 then the Initial Layer Speeds will be what you enter in the box.  If your print speed is 40 but you have entered 30 as the Initial Layer Speed then the first layer will be at 30.
    - If you have "Number of Slower Layers" set above 1 then the intermediate layers will have graduated speeds from your Initial Layer Speed until they get to Print Speeds for the individual features.
  - The units for all accel settings are in mm/sec²
  - The units for Jerk are in mm/sec.
 - If you use something like "Change At Z" then it uses M220 to tell the printer to adjust all speeds it sees by the percentage.  M220 S50 would tell the printer that all speeds are to be calculated to 50%.  It is exactly the same thing when you adjust the "Feedrate" on the LCD.  All speeds are calculated to the % of your setting.  It is in fact an internal M220 for the mainboard.
 - If Cura notes that a layer is less than "Minimum Layer Time" it will adjust the printing speed downward to maintain "Minimum Layer Time".  It will adjust down to whatever you have for "Minimum Layer Speed".
 - If you have "Flow Equalization Ration" at 100%, and Cura calculates an extrusion at (for example) 1/2 your set line width then the speed of that extrusion will be faster in order to maintain a constant flow rate through the nozzle.

In the gcode, all speeds are in mm/minute.  Dividing any "F" value by 60 will convert to mm/sec.

kevinjones4642, 2024-04-08:
Thanks for the details, but I'm still lost why my setting of 40mm/s results in speeds topping in Cura 5.6 at 30mm/s, and in Cura 5.7 topping at 51.56 mm/s. In both cases, it appears to be more of an interpretation than an actual command. Also, if I set print speed to something less than default, do all of the sub-speeds get scaled accordingly?

I'm also trying to figure out why the default setting for PETG with the 0.25 nozzle jumped from 55 mm/s to 100 mm/s going from 5.6 to 5.7.

kevinjones4642, 2024-04-08:
Digging into the speed settings further, both the profiles and the behavior are very different. In 5.6, the default speed for PETG and 0.25 nozzle is 55 mm/s. If I change "Print Speed" to 40 mm/s, and look at what happens to the 20 sub-speed values, the highest is actually 30mm/s. First layer and brim are both 7.5 mm/s. In 5.7, the default speed is 100 mm/s, and my setting of 40 mm/s results in many of the sub speeds actually being at 40 mm/s. Notable, the 1st layer and brim are at 30 mm/s, 4 times faster. 

Oddly, when I preview the sliced object, if I view speeds, in 5.6 I can see that the reported sub-speeds are actually correctly enforced, but in 5.7, even though the peak speed for any printing action was 40 mm/s, it is actually printing most of the interior at 51.56 mm/s. At the other end, outer walls are supposed to print at 8 mm/s, but are shown as being 3.97 mm/s. I can't believe this is intentional.

kevinjones4642, 2024-04-08:
Sorry for the spam, but there seems to be some strange disconnect between what Cura shows as the speed, and what it actually does. In the extended settings there is a master speed and 20 sub-speeds. In 5.7, if I change the master speed, say from 100 mm/s to 40 mm/s, all of the sub-speeds shown in the printer settings reflect scaled values. If I slice, and then view the commanded speeds, none of them are remotely close to what is shown in the settings.

GregValiant, 2024-04-09:
Set Cura up to slice using a model that shows the problem.  Use the "File | Save Project" command and then zip the resultant 3mf file and post the zip folder here.

If you have "Flow Compensation Ratio" at 100% then the speed of any extrusion can vary according to the line width.  Usually the outside walls are not affected but inside walls are a different story.  Setting the Flow Compensation to 0 will result in no speed adjustments regardless of the flow rate through the nozzle.

_"I'm also trying to figure out why the default setting for PETG with the 0.25 nozzle jumped from 55 mm/s to 100 mm/s going from 5.6 to 5.7."_
I can't duplicate that either.  It might have something to do with the S5 definition file.  UltiMaker does a lot of testing with materials for their printers.  Consequently a lot of people are involved and the chances of a typo in there someplace goes up.
If you right click on a setting box an option will to "Use value from material".  Maybe that will show something?

kevinjones4642, 2024-04-09:
Thanks for the feedback. There is a zip file attached to the 1st post in the thread with the project file shown in the pictures above. I've been playing around a lot in Cura trying to understand what it is doing, speed-wise, but still haven't decoded the logic. For the time being I have rolled back to 5.6, which works well.

There are definitely some major changes in profile settings. As you say, possibly typos. There are other things I don't quite understand. In 5.6, the "Print Speed" setting scaled of the sub-speeds accordingly, and the actual sliced file appeared to honor those speeds. In 5.7, the reported min and max print speeds cover a much wider range than the min and max speeds shown in the Print Settings. I guess I'll wait for the next update on 5.7 before I try it again.