Ultimaker/Cura #18835: failure to download if model left in center of buildplate on file open

moviebrain, 2024-04-07:
### Cura Version

5.7.0

### Operating System

windows 10

### Printer

wanhao i3

### Name abnormal settings

none, tpu profile .3 first layer, .16 all others.

### Describe model location

center of build plate.

### Describe your model

flexi rex, old save and newly downloaded. mesh fix tool found it watertight.

### Add your .zip here ⬇️

[WDI3_Flexi-Rex-improved.zip](https://github.com/Ultimaker/Cura/files/14896046/WDI3_Flexi-Rex-improved.zip)

only slices if moved from center of buildplate. I have had nothing but slicing errors over and over since updating to 5.6 and now 5.7, items have have always sliced error-free before.

GregValiant, 2024-04-07:
Thanks for the report.
5.6.0 had definite problems slicing some models.  I didn't have any trouble slicing this in 5.7.0 which all around appears to be much better.
Use the "Help | Show Configuration Folder" command to find the Cura.log file.  Zip that up and post the zip folder here.  There might be a clue in the log as to what is going on.

moviebrain, 2024-04-07:
2024-04-07 06:46:11,142 - DEBUG - [zeroconf-ServiceBrowser-_octoprint._tcp-21224] OctoPrintPlugin.OctoPrintOutputDevicePlugin._onServiceChanged [416]: Bonjour service added: octopi._octoprint._tcp.local
2024-04-07 06:46:23,531 - INFO - [MainThread] cura.CuraApplication.resetAllTranslation [1494]: Resetting all scene translations
2024-04-07 06:46:23,542 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: GroupedOp.(#=1) SetTransformOp.(node=<CuraSceneNode object: 'Flexi-Rex-improved.stl'>), took 9ms
2024-04-07 06:46:24,971 - INFO - [MainThread] CuraEngineBackend.CuraEngineBackend._createSnapshot [331]: Creating thumbnail image (just before slice)...
2024-04-07 06:46:24,972 - DEBUG - [MainThread] UM.Scene.Camera._preferencesChanged [255]: Changing perspective mode to perspective.
2024-04-07 06:46:24,974 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [61]: Loading [C:\Program Files\UltiMaker Cura 5.7.0\share\cura\resources\shaders\overhang.shader]...
2024-04-07 06:46:24,976 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [61]: Loading [C:\Program Files\UltiMaker Cura 5.7.0\share\cura\resources\shaders\transparent_object.shader]...
2024-04-07 06:46:24,982 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [61]: Loading [C:\Program Files\UltiMaker Cura 5.7.0\share\cura\resources\shaders\striped.shader]...
2024-04-07 06:46:25,041 - INFO - [MainThread] CuraEngineBackend.CuraEngineBackend.slice [348]: Starting to slice...
2024-04-07 06:46:25,041 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend.slice [368]: Going to slice build plate [0]!
2024-04-07 06:46:25,042 - DEBUG - [MainThread] UM.Backend.Backend._createSocket [294]: Previous socket existed. Closing that first.
2024-04-07 06:46:25,229 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [232]: Socket state changed to Listening
2024-04-07 06:46:25,260 - INFO - [MainThread] UM.Backend.Backend.startEngine [96]: Started engine process: C:\Program Files\UltiMaker Cura 5.7.0\CuraEngine.exe
2024-04-07 06:46:25,265 - DEBUG - [MainThread] UM.Backend.Backend._backendLog [140]: [Backend] Calling engine with: ['C:\\Program Files\\UltiMaker Cura 5.7.0\\CuraEngine.exe', 'connect', '127.0.0.1:49674', '']
2024-04-07 06:46:25,324 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2024-04-07 06:46:25.323] [info] Sentry url: https://734f9ec9024f73e53701d59c3ffddfe3@o323038.ingest.sentry.io/4506257745510401
2024-04-07 06:46:25,333 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2024-04-07 06:46:25.323] [info] Sentry config path: C:/Users/*******/AppData/Roaming/cura/.sentry-native
2024-04-07 06:46:25,337 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [236]: Socket state changed to Connected
2024-04-07 06:46:25,339 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2024-04-07 06:46:25.323] [info] Starting sentry
2024-04-07 06:46:25,340 - DEBUG - [MainThread] UM.Backend.Backend._onSocketStateChanged [225]: Backend connected on port 49674
2024-04-07 06:46:25,340 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2024-04-07 06:46:25.334] [info] Connecting to 127.0.0.1:49674
2024-04-07 06:46:25,371 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onStartSliceCompleted [580]: Sending slice message took 0.329120397567749 seconds
2024-04-07 06:46:25,440 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2024-04-07 06:46:25.440] [info] Connected to 127.0.0.1:49674
2024-04-07 06:46:25,611 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2024-04-07 06:46:25.610] [info] All settings:  -s infill_extruder_nr="-1" -s bridge_sparse_infill_max_density="0" -s quality_changes_name="WDI3_Flexi-Rex-improved(1) TPU" -s speed_wall_x="60.0" -s prime_tower_wipe_enabled="True" -s support_offset="0.8400000000000001" -s wall_0_material_flow_roofing="100" -s support_roof_pattern="concentric" -s support_interface_height="1" -s brim_gap="0" -s speed="0" -s zig_zaggify_infill="False" -s support_roof_angles="[ ]" -s cool_min_layer_time_fan_speed_max="10" -s prime_blob_enable="False" -s support_roof_density="100" -s bridge_skin_support_threshold="50" -s anti_overhang_mesh="False" -s support_roof_wall_count="0" -s machine_max_acceleration_e="10000" -s support_bottom_stair_step_width="5.0" -s material_crystallinity="False" -s machine_name="Wanhao Duplicator i3" -s support_brim_width="1.98" -s support_xy_distance="0.5" -s arcwelder_maximum_radius="9999" -s material_brand="empty_brand" -s extruder_prime_pos_x="0" -s support_interface_density="100" -s nozzle_disallowed_areas="[]" -s wipe_retraction_amount="6.5" -s prime_tower_max_bridging_distance="5" -s support_tower_maximum_supported_diameter="3.0" -s wall_overhang_angle="90" -s machine_endstop_positive_direction_y="False" -s skin_material_flow_layer_0="100" -s wipe_hop_enable="False" -s magic_fuzzy_skin_enabled="False" -s min_even_wall_line_width="0.34" -s cross_infill_density_image="" -s switch_extruder_extra_prime_amount="0" -s arcwelder_extrusion_rate_variance="5" -s jerk_print="20" -s build_volume_temperature="0" -s infill_support_enabled="False" -s draft_shield_dist="10" -s support_meshes_present="False" -s cool_fan_full_layer="2" -s _plugin__curaenginegradualflow__0_1_0__reset_flow_duration="2.0" -s material_flush_purge_length="60" -s speed_support_interface="10.0" -s brim_width="8.0" -s machine_nozzle_head_dista [... cut ...]
2024-04-07 06:46:28,550 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2024-04-07 06:46:28.550] [info] Connecting to 127.0.0.1:49674
2024-04-07 06:46:28,551 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [236]: Socket state changed to Connected
2024-04-07 06:46:28,552 - DEBUG - [MainThread] UM.Backend.Backend._onSocketStateChanged [225]: Backend connected on port 49674
2024-04-07 06:46:28,658 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2024-04-07 06:46:28.658] [info] Connected to 127.0.0.1:49674

moviebrain, 2024-04-07:
This was Cura 5.7.0, and is completely repeatable by "reset all model positions" and slicing with the options provided in the project and above. Why it then slices when moved just a little bit I couldn't tell you.

GregValiant, 2024-04-07:
@HellAholic 
This is the first time I've this return code:  3221226505

HellAholic, 2024-04-07:
@moviebrain Try removing/disabling plugins and attempt to slice again. I can slice in the middle of the build plate on my personal device. It has no plugins and no custom settings/profiles. Just opened your project file and pressed slice. Also tried with reset all model positions, no slice fail. 
[image]
Let me know if that solves the issue.

moviebrain, 2024-04-07:
I went into marketplace and found 3 non-bundled extensions (arc welder and mesh tools, both unused for this slice, and octoprint connection manager). I disabled all three and restarted cura, the slice still fails with exit code 3221226505 still.

HellAholic, 2024-04-07:
Thanks for checking, I'll have to run it on another device to see if the crash is reproducible there.
Outside of that I'll see if the sentry shows more detail for the slice fail.

moviebrain, 2024-04-07:
I just went back in to test something, if opening the cura project on my own machine changed the workflow somehow and allowed the slice to proceed. 

It did not. It still fails for me unless I move the model a bit. 

It’s frustrating because this isn’t a huge issue at the end of the day, but also it hasn’t always been a problem, this has worked without jiggling stuff in the not so distant past. 

[image]

HellAholic, 2024-04-08:
@moviebrain Thanks for additional info and going into detail with the investigation. I've added a ticket for the team with your pointers/findings.

For internal reference of the Cura team CURA-11821