Ultimaker/Cura #18837: Error when opening 3mf files containing several aligned objects.

Blackmoon85, 2024-04-07:
### Cura Version

5.7.0

### Operating System

Windows 11

### Printer

Kywoo3D / Tycoon IDEX

### Reproduction steps

Problem with Cura 5.7.0, when opening a 3MF file (from an export from Fusion360 where several bodies are correctly aligned with each other) the main body is automatically opened in the center of the pallet with the various other bodies spread all around it and automatically lowered onto the tray instead of all remaining aligned as in Fusion360. Whereas under Cura 5.6.0 the bodies opened at the origin of the platen and were correctly aligned with each other, as under Fusion360.
The "Automatically lower models to tray" option in Preferences/Configure Cura... is indeed disabled in both versions.

Is this due to a bug or is it a new option I haven't seen?

### Actual results

All objects are centered on the tray and automatically lowered onto the tray.

### Expected results

Objects remain aligned with each other.

### Add your .zip and screenshots here ⬇️

[image]
[image]
[image]
[image]

GregValiant, 2024-04-07:
Please provide a project file.  Without one there is just no way to tell what is going on.
Have you tried using 'Select All Models" and then "Merge Models"?

Blackmoon85, 2024-04-07:
Hello,

In Cura 5.6, the objects from my Fusion360 3mf exports are displayed at the
tray origin and to bring them back to the tray, I do the following, "Select
all models" then "Group models" then "Center selection" and finally
"Ungroup models", so afterwards I can select the models to assign them the
print extruder according to its color. The various objects in the template
are still aligned with each other as expected.

Whereas with Cura 5.7, as soon as I open the export, it's placed directly
in the center of the tray, with all the objects making up the model
automatically lowered to the level of the tray. It's as if there were an
automatic command to center the objects as they open.

Best regards

Blackmoon85, 2024-04-07:
Oops, I forgot the 3mf file as an example.

bioxz, 2024-04-07:
I think I'm affected by the same issue. With 5.6 and after disabling the "Ensure models are kept apart", opening a 3mf project would keep everything in the original position. With 5.7, it seems like the feature is always active, thus multi part models get always ripped apart and placed next to each other separately.

Example file: https://recyclingfabrik.com/wp-content/uploads/2024/01/Recycling_Fabrik_MultiSpool_V1.0.zip

This example has a spool ring and separate spool walls, the spools walls are position in the 3mf exactly where they should print. In 5.6 the files opened as expected, with 5.7 the model parts always get separated and it's nigh impossible to reassemble to correct full model. 

The mentioned "Merge Models" function just places all selected parts centered, not as it was placed as in the 3mf.

Blackmoon85, 2024-04-07:
Hi,

Indeed, it could be a lead. So I tried this setting with Cura 5.6 by
activating it. When I opened the 3mf file, the model still opened at the
origin with all objects correctly aligned. However, after selecting all the
objects, grouping them and centering them on the tray by ungrouping them,
the objects broke up and broke the alignment. However, this is not the same
effect I get with Cura 5.7, because not only are the objects not lowered to
the level of the tray, but this produces a kind of cloud with the smaller
objects, with the largest and main object being pushed aside, whereas with
Cura 5.7 the smaller objects are all around the main one.

I've also checked Cura 5.7 and the parameter is indeed deactivated.

It's really as if by default Cura 5.7 opened the 3mf with centering and the
parameters "Automatically lower models on the tray" and "Keep models
separated" enabled, whereas in Cura's configuration this is not the case
(under Cura 5.6 with these 2 parameters enabled I still don't get exactly
the same result).