Ultimaker/Cura #18839: Add Ender-3 V3 to the list of supported printers

JMH-Jarbydev, 2024-04-07:
### Is your feature request related to a problem?

I would like to have the Ender-3 V3 CoreXY added to the list of available printers.

https://store.creality.com/products/ender-3-v3-corexz-3d-printer?spm=..index.album_1.1&spm_prev=..product_fafd7f3c-344b-4298-b2dd-7e2a361430ad.breadcrumbs_1.1

### Describe the solution you'd like

I would like to have the Ender-3 V3 CoreXY added to the list of available printers.

https://store.creality.com/products/ender-3-v3-corexz-3d-printer?spm=..index.album_1.1&spm_prev=..product_fafd7f3c-344b-4298-b2dd-7e2a361430ad.breadcrumbs_1.1

### Describe alternatives you've considered

I tried to use similar Printers but didn't work.

### Affected users and/or printers

Creality  Ender-3 V3 CoreXY added to the list of available printers.

GregValiant, 2024-04-07:
This has to come from Creality or somebody who owns one.  I see the "Ender 3 V3 SE" is supported.  What's different?
If it is supported in Creality Cura you could possibly copy the definition file, extruder files, and nozzle files from that app to UM Cura.

hphpc, 2024-06-01:
Hi,

I own Creality Ender-3 V3. However, I do not have any experience of prior V3 minor versions (SE & KE). In response to previous query, please note the following differences between the two I could figure out (a lot of that info comes from: https://www.crealityexperts.com/creality-ender-3-v3-vs-ender-3-v3-se)

### **Motion System:**
**Ender-3 V3**: Utilizes a CoreXZ motion system, which helps reduce vibration and improves print quality. This system is designed to enhance stability and accuracy by using linear rods on all axes​.
**Ender-3 V3 SE**: Uses a traditional Cartesian motion system. While it includes improvements like linear rods for the Y-axis, it doesn't offer the advanced stability of the CoreXZ system

### **Print Speed:**
**Ender-3 V3**: Capable of reaching up to 600 mm/s with an acceleration of 20,000 mm/s², making it one of the fastest in its class.
**Ender-3 V3 SE**: Can reach a maximum of 250 mm/s with an acceleration of 2,500 mm/s². This is slower compared to the Ender-3 V3 but still faster than many older models

### **Hot-end and Nozzle:**
**Ender-3 V3**: Features a newer hot-end design capable of reaching 300°C, allowing it to print high-temperature materials such as nylon. It also has a quick-change nozzle design for easier maintenance
**Ender-3 V3 SE**: Equipped with the Sprite hot-end capable of reaching 260°C. This limits it to more common filaments like PLA, ABS, and PETG

### **Build Surface:**
**Ender-3 V3**: Comes with a textured PEI build plate, which offers excellent adhesion and ease of part removal​.
**Ender-3 V3 SE**: Uses a PC spring steel bed, which provides good adhesion but is not as durable or user-friendly as the PEI plate on the V3​ 

### **Firmware and Connectivity:**
**Ender-3 V3**: Runs on Klipper-based firmware, providing better control algorithms for improved print quality. It supports WiFi connectivity for wireless operation and remote monitoring.
**Ender-3 V3 SE**: Uses more basic firmware with SD card and USB connectivity. It lacks the advanced features and WiFi capability of the V3​

### **Display:** 
**Ender-3 V3**: Includes a 4.3” color touchscreen for a more intuitive user interface​.
**Ender-3 V3 SE**: Features a 3.2” color LCD screen with a knob controller, which is simpler but less modern compared to the V3’s touchscreen

### **Conclusion:**
The Ender-3 V3 is a more advanced and capable printer, with a higher maximum print speed, better hot-end and nozzle capabilities, more advanced firmware, and better connectivity options. The Ender-3 V3 SE, while still a strong performer, is more of an entry-level option with fewer advanced features.

Disclaimer: I merely used a comparative analysis tool to pull some details out, if anything is misrepresented here, please do double check.

**I also would like to request this profile to be added in the next version upgrade of Cura.**

Thanks

hphpc, 2024-06-01:
Also I stumbled across the following - that may be helpful
https://elopage.com/s/the3dprinterbee/cura-profiles?locale=en
https://www.youtube.com/watch?v=fsTHNXIbSng

sicko666, 2024-07-08:
I am adding my request Cura for profiles for this printer.
Am just a hobbiest who loves printing stuff out to paint, but had this same issue when i had my flashforge creator 2 Duo printer. Took ages for any profiles, most of the time had to use a single extruder profile because no profiles for two creator extruders existed! Now i have a Creality, thought would there be some, as its a Cura based app that Creality uses, but no profiles for it in Cura. I know there will be eventually, am just putting in the request so experts who know how to do it, know theres a few of us out there that have no idea how to do these complicated things, we just wanna create!

GregValiant, 2024-07-08:
When someone writes a definition and puts it and any supporting files together in a Pull Request it will be considered for inclusion in Cura.  I don't own one so I know nothing about them.  UltiMaker isn't going to pay their employees to support a 3rd party printer.
Definitions must come from the community or the manufacturer.  I haven't seen Creality submit any definitions in a while so I assume they want you to use the Creality slicer.
If the V3 definition files are included with the Creality slicer then they may, or may not, work with UltiMaker Cura.  They would need to be carefully checked to insure that the proper settings are included.  Most will be fine but there are new settings added and obsolete settings deleted with each UM Cura release.  Those will not be supported in the Creality definitions and the Creality definitions are based on a Cura version that is around 17 versions behind UM Cura.

sicko666, 2024-07-18:
Oh thank you, explains a lot. Dont get me wrong, i love open source, its the printer companies i am fed up with. BOTH printers claimed to be open source compatible, hence i got them, because of that promise. Yet with both, no profiles existed in the open source communities. The companies did not release the profiles, so they lied when they said open source compatible! Its that what annoys me, the blatent, on purpose, lie by these companies! I will NEVER be disapointed with open source material, its created by individuals, by communities trying to help each other, i love that fact! Thats why i always try, when ever possible, to use open source. Hence my desire to use Unimaker Cura over Crealities version of Cura, i do not even understand Why theres a Creality version of Cura, why not just use Cura and release the profiles for their print to them? To me, thats logical, creating another version of something thats already there, an fully supporteed already, thats ilogical! Like you say 17 versions behind already an this version of the printer was only released this year! Lol hence my desire, as well as others i can imagine, to use Ultimaker Cura with profiles for our printers over a version of Cura thats 17 times behind the current version! ;)

GregValiant, 2024-07-18:
Creality, Sovol, LottMaxx, Mingda, Wanhao, and others, each has their own fork of Cura.  I have no idea why each company went that way.  In the last year or two I have noticed that Creality has stopped submitting printer definitions to UM Cura and simply adds them to their own version.
I suppose part of the reason for "proprietary forks" is similar to why UM developed it in the first place.  Your own slicer can be optimized for your own line of printers.

That's all well and good, but each fork needs to be maintained.  That maintenance causes divergence between the succeeding versions of any company's fork, and between the main fork of each company.  The bottom line is that just because (for example) Creality Cura has definitions for your printer does not mean that those definitions can simply be copied over to UM Cura, or some other fork of Cura.  They might work, but they might not.  Compatibility can fall by the wayside.
UM Cura 5.8-beta has 654 settings.  For comparison;  UM Cura 4.13 has 558 settings and I would suspect that Creality 4.8.2 has a few less.  Either way, that's a big difference and some of those differences will show up in definition files.

The alternative to each company having and maintaining it's own fork is to have a "Universal" fork.  I have opted for UM Cura, and for the SmartAvionics fork, as they are well maintained and easily adaptable (customizable) to the way I want my Ender to print.  Like most of those other forks, the SmartAvionics forks is built on Cura 4.x so is pre-Arachne.  Since there is no way to turn off Arachne in UM Cura, I use SmartAvionics Master when I don't want or require variable line width and for it's Spiralize ability which is really good.  In addition, SA Master is close enough to allow me to use my post-processors without any re-writes to adapt them from one version to the other.

hphpc, 2024-07-18:
Sorry how do these last few comments help in speeding up Cura support for the printer model Ender 3 V3?

Keep comments to constructive level. I can go on about countering all the arguments/statements above, however this is not Reddit or Pinterest please.

hphpc, 2024-07-18:
> This has to come from Creality or somebody who owns one.  I see the "Ender 3 V3 SE" is supported.  What's different?
> 
> If it is supported in Creality Cura you could possibly copy the definition file, extruder files, and nozzle files from that app to UM Cura. 

I am not sure about this copying part. Maybe I'll try myself to see what's the fuss about. If you were to ask me to type in a DAG using raw C pointers might have easier, since I am new to this 3d printing world. I'll try and if any success will throw in the configs here (if possible).

GregValiant, 2024-07-19:
Someone has to submit the definition files to UltiMaker for inclusion in Cura.  Then the UltiMaker team reviews the submission for the "form" but not for the values.  There is now way to check if the values are correct. 
If it is deemed to be written correctly (no obvious errors in the code), it would be included in the next release of Cura.
It isn't possible for the UltiMaker Company to write definition files for machines manufactured by someone else.  

What you can do right now is:
- Install the SE model and during installation (or afterword) change the "printer name".
- Go to Settings / Manage Printers / Machine Settings and double check that the build volume (X(width) x Y(depth) x Z(height)) and other physical things about the printer are set correctly.
- Go to the Cura "MarketPlace" and load the "Printer Settings" plugin.
- Scroll down and set the "Maximum" values of "Acceleration" and "Speeds" to match your printer.  Review the rest of the Printer Settings and make sure they are correct for your V3.
- The most important setting is "G-Code Flavor" in the Machine Settings.  Second most important might be "Retraction Distance" in the Travel settings.

That should get you started.  There are support groups on [Reddit ](https://www.reddit.com/r/Ender3V3/)where you can ask regarding other setting changes that might be required.