Ultimaker/Cura #18843: Segmentation fault when opening file dialog

callistix, 2024-04-07:
### Cura Version

5.7.0

### Operating System

Debian Linux 12.4

### Printer

Creality Ender v2

### Reproduction steps

When running the `UltiMaker-Cura-5.7.0-linux-X64.AppImage` Cura starts normally, although there is a warning in the terminal:

```
qt.qml.typeresolution.cycle: Cyclic dependency detected between "file:///tmp/.mount_curaMfNxNE/share/cura/resources/qml/Actions.qml" and "file:///tmp/.mount_curaMfNxNE/share/cura/resources/qml/Actions.qml"
```

Everything seems to work normally, I've successfully set up my printer, the filament and other preferences.
But each time I select "File -> Open File(s)", the application crashes with `Segmentation fault` without further information. It never gets to the point where it is supposed to show a file selection dialogue.

I've tried with older versions (5.4.0) and it also crashes.

The `strace` output doesn´t seem to be too useful either:

```
write(7, "scipy.ndimage._ni_label", 23) = 23
write(7, ", ", 2)                       = 2
write(7, "pySavitar", 9)                = 9
write(7, " (total: ", 9)                = 9
write(7, "88", 2)                       = 2
write(7, ")", 1)                        = 1
write(7, "\n", 1)                       = 1
gettid()                                = 37001
getpid()                                = 37001
tgkill(37001, 37001, SIGSEGV)           = 0
--- SIGSEGV {si_signo=SIGSEGV, si_code=SI_TKILL, si_pid=37001, si_uid=1000} ---
+++ killed by SIGSEGV +++
Segmentation fault
```

Could you please assist? Thanks.

### Actual results

Cura crashes with a Segmentation fault.

### Expected results

Cura should show a file selector dialog.

### Add your .zip and screenshots here ⬇️

none

xptian, 2024-05-10:
Currently getting this same bug on Cura 5.7.1 when using the AppImage on Arch Linux.

`./UltiMaker-Cura-5.7.1-linux-X64.AppImage 
qt.qml.typeresolution.cycle: Cyclic dependency detected between "file:///tmp/.mount_UltiMagmhNC9/share/cura/resources/qml/Actions.qml" and "file:///tmp/.mount_UltiMagmhNC9/share/cura/resources/qml/Actions.qml"
Segmentation fault (core dumped)`

theLogicJB, 2024-05-27:
I get the same behavior and the same console output. But I got some more `strace` output. This is everything that is printed after cura started and I clicked the button on the main screen to open a file.

```
0x7ffd2aec72b0, 0, NULL)      = ? ERESTARTSYS (To be restarted if SA_RESTART is set)
--- SIGWINCH {si_signo=SIGWINCH, si_code=SI_KERNEL} ---
wait4(-1, [{WIFSIGNALED(s) && WTERMSIG(s) == SIGSEGV && WCOREDUMP(s)}], 0, NULL) = 1144406
rt_sigaction(SIGINT, {sa_handler=SIG_DFL, sa_mask=[], sa_flags=SA_RESTORER, sa_restorer=0x7523f73ed770}, {sa_handler=0x63e2193761d0, sa_mask=[], sa_flags=SA_RESTORER, sa_restorer=0x7523f73ed770}, 8) = 0
ioctl(2, TIOCGWINSZ, {ws_row=76, ws_col=319, ws_xpixel=2552, ws_ypixel=1368}) = 0
openat(AT_FDCWD, "/usr/share/locale/locale.alias", O_RDONLY|O_CLOEXEC) = 4
fstat(4, {st_mode=S_IFREG|0644, st_size=2998, ...}) = 0
read(4, "# Locale name alias data base.\n#"..., 4096) = 2998
read(4, "", 4096)                       = 0
close(4)                                = 0
openat(AT_FDCWD, "/usr/share/locale/en_GB.UTF-8/LC_MESSAGES/bash.mo", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/share/locale/en_GB.utf8/LC_MESSAGES/bash.mo", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/share/locale/en_GB/LC_MESSAGES/bash.mo", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/share/locale/en.UTF-8/LC_MESSAGES/bash.mo", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/share/locale/en.utf8/LC_MESSAGES/bash.mo", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/share/locale/en/LC_MESSAGES/bash.mo", O_RDONLY) = -1 ENOENT (No such file or directory)
fstat(2, {st_mode=S_IFCHR|0620, st_rdev=makedev(0x88, 0xf), ...}) = 0
openat(AT_FDCWD, "/usr/share/locale/en_GB.UTF-8/LC_MESSAGES/libc.mo", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/share/locale/en_GB.utf8/LC_MESSAGES/libc.mo", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/share/locale/en_GB/LC_MESSAGES/libc.mo", O_RDONLY) = 4
fstat(4, {st_mode=S_IFREG|0644, st_size=1430, ...}) = 0
mmap(NULL, 1430, PROT_READ, MAP_PRIVATE, 4, 0) = 0x7523f7630000
close(4)                                = 0
openat(AT_FDCWD, "/usr/share/locale/en.UTF-8/LC_MESSAGES/libc.mo", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/share/locale/en.utf8/LC_MESSAGES/libc.mo", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/share/locale/en/LC_MESSAGES/libc.mo", O_RDONLY) = -1 ENOENT (No such file or directory)
write(2, "/usr/bin/cura: line 4: 1144406 S"..., 101/usr/bin/cura: line 4: 1144406 Segmentation fault      (core dumped) /opt/ultimaker-cura/AppRun "$@"
) = 101
rt_sigprocmask(SIG_SETMASK, [], NULL, 8) = 0
--- SIGCHLD {si_signo=SIGCHLD, si_code=CLD_DUMPED, si_pid=1144406, si_uid=1000, si_status=SIGSEGV, si_utime=593 /* 5.93 s */, si_stime=257 /* 2.57 s */} ---
wait4(-1, 0x7ffd2aec6b50, WNOHANG, NULL) = -1 ECHILD (No child processes)
rt_sigreturn({mask=[]})                 = 0
read(255, "", 77)                       = 0
rt_sigprocmask(SIG_BLOCK, [CHLD], [], 8) = 0
rt_sigprocmask(SIG_SETMASK, [], NULL, 8) = 0
exit_group(139)                         = ?
+++ exited with 139 +++
```

mw75, 2024-06-02:
+1 5.7.0 and 5.7.1 bullseye/sid

Estebiu, 2024-06-14:
Same thing has happened to me on arch for the last 2/3 months. Strangely, on SteamOS it runs just fine.

WaviestBalloon, 2024-07-06:
Having the exact same issue here as well on the latest version of Cura.

```
qt.qml.typeresolution.cycle: Cyclic dependency detected between "file:///opt/ultimaker-cura/share/cura/resources/qml/Actions.qml" and "file:///opt/ultimaker-cura/share/cura/resources/qml/Actions.qml"
/usr/bin/cura: line 11: 1414901 Segmentation fault      (core dumped) /opt/ultimaker-cura/AppRun "$@"
```

Running on Arch Linux, `6.9.5-zen1-1-zen`

42Willow, 2024-07-27:
This is a duplicate of #16815 I think

KLudwig-V, 2024-09-30:
Still same issue with 5.8.1

mbenkmann, 2024-11-26:
Still exists with Cura 5.9.  I just installed a fresh Ubuntu 24.04. I still have my old Ubuntu 20.04 that I can boot into. I can confirm that the same .AppImage crashed on Ubuntu 24.04 but doesn't crash on Ubuntu 20.04.

mbenkmann, 2024-11-26:
Thank you @42Willow for posting the reference to the old bug. It contains a workaround. Force it to use the GTK3 platform theme:

    UltiMaker-Cura-5.9.0-linux-X64.AppImage -platformtheme gtk3

I'd like to note that on my old Ubuntu 20.04 Cura uses the GTK file dialogs even without this switch and doesn't crash. So it seems to be related to using the Qt file dialogs.

WaviestBalloon, 2024-11-26:
For me, the issue has subsided on Arch Linux, now it just doesn't want to open anything even via drag-and-drop. Sometimes it does but it takes over 5 minutes for it to register that a file was selected.

roboman2444, 2025-09-13:
Can confirm this still happens on Cura 5.10.2 on arch (latest installed via cura-bin in AUR).
-platformtheme gtk3 is a workaround.