Ultimaker/Cura #18845: Nose

Danrathburn321, 2024-04-07:
### Cura Version

5.6

### Operating System

Win 11

### Printer

Antcubic max

### Name abnormal settings

none. Same as always

### Describe model location

Center flat

### Describe your model

AC landing gear

### Add your .zip here ⬇️

[Nose.zip](https://github.com/Ultimaker/Cura/files/14899260/Nose.zip)

Danrathburn321, 2024-04-07:
has failed on last three tries of same type file

HellAholic, 2024-04-08:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well?
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

If you did not manage to successfully slice, please include a project file to allow us to check with your printer settings and profile changes.
Providing the log also helps with troubleshooting. Use the "Help | Show Configuration Folder" on the top bar to find the Cura.log file. Zip that up and post the zip folder here.

closed 2024-04-22 by github-actions[bot]