Ultimaker/Cura #18846: Slicing failed on all files - tried debugging myself

johnneerdael, 2024-04-08:
### Cura Version

5.7.0 / 5.6.0

### Operating System

Windows 11

### Printer

FLSUN V400

### Name abnormal settings

Slicing fails with custom and stock settings/profiles

### Describe model location

No changes do not affect slicing

### Describe your model

Yes, in older cura versions

### Add your .zip here ⬇️

[cura.zip](https://github.com/Ultimaker/Cura/files/14899495/cura.zip)
File contains whole config folder as well and error files

johnneerdael, 2024-04-08:
Already disabled most plugins and even some stock ones that showed errors. Trying to rollback to 5.5.0 to see if that works.

johnneerdael, 2024-04-08:
I can confirm that slicing in 5.5 works without issues.

HellAholic, 2024-04-08:
Hey @johnneerdael ,
I looked at the log files you've provided, the error seems to be caused by missing a setting `prime_tower_enable="False"` which causes the error:
`[error] Trying to retrieve setting with no value given: prime_tower_mode`

Could you check a few steps to see if the issue will be resolved in 5.6/5.7?

- Close all cura versions (if any)
- Go to the configuration folder
- Delete the 5.6 and 5.7 folders but keep the 5.5 working version
- Start Cura 5.7
- Load the project and slice

If the above steps resolved the issue:
- Could you please specify how you added the profiles for your printer?

There might be a upgrade script step missing, hence my question.
Note that I can slice the project using your configuration and model, so I'm wondering what steps I'm missing or could be the cause.

Cheers!

closed 2024-04-22 by github-actions[bot]