Ultimaker/Cura #18853: No Temp After Heating and Priming

LowtechEnvy, 2024-04-08:
### Cura Version

5.7.0

### Operating System

Windows 10 Home

### Printer

Ender 5 Marlin

### Reproduction steps

Sliced and put in printer

### Actual results

Printer heated - Primed - E1 and bed heat shut off

### Expected results

Resliced and continued to happen in 5.7.0 - sliced in 5.50 and no issues

### Add your .zip and screenshots here ⬇️

[CE5_xyzCalibration_cubebad.zip](https://github.com/Ultimaker/Cura/files/14910398/CE5_xyzCalibration_cubebad.zip)

HellAholic, 2024-04-08:
You've made mistakes in your custom Start G-Code.
That code will not run correctly in 5.5.0, 5.6.0, or 5.7.0. So I'm confused how you've got it working in 5.5.0.

`M140 S{first_layer_bed_temperature[0]} ; set final bed temp
M104 S{first_layer_temperature[0]} ; set final nozzle temp

M190 S{first_layer_bed_temperature[0]} ; wait for bed temp to stabilize
M109 S{first_layer_temperature[0]} ; wait for nozzle temp to stabilize`

5.5.0 Results in:

`M140 S{ ; set final bed temp
M104 S{ ; set final nozzle temp

M190 S{ ; wait for bed temp to stabilize
M109 S{ ; wait for nozzle temp to stabilize`

And in 5.7.0 it actually resolves into 0:

`M140 S0 ; set final bed temp
M104 S0 ; set final nozzle temp

M190 S0 ; wait for bed temp to stabilize
M109 S0 ; wait for nozzle temp to stabilize`

LowtechEnvy, 2024-04-08:
Nothing I have changed in my start code.  Its odd that it is a customer G
code start.  It has been the same from Cura 3 or 4.  I will check it out.

Thanks,

Brennan

GregValiant, 2024-04-08:
Did you open a project file from a customer?  It would over-write the StartUp Gcode of your printer.

In the Machine Settings in Cura the printer is a single extruder machine, but the "Fan Number" is set to 1.  Normally it is 0.  The setting isn't "how many fans" but rather what fan circuit is involved as the layer cooling fan for a particular extruder.

The "keywords" being used in the StartUp Gcode are from PrusaSlicer.  When the Cura sees them instead of swapping out the keywords and putting in the temperature it is getting stuck.  Then when the printer sees the lines that Hellaholic is showing it simply turns off the bed and hot end.

M140 S{first_layer_bed_temperature[0]} ; set final bed temp       ;This should be **{material_bed_temperature_layer_0**}
M104 S{first_layer_temperature[0]} ; set final nozzle temp   ; should be {**material_print_temperature_layer_0**}

Does the printer have an ABL?  M420 S1 is setting up the ABL but Creality printers often use G29 as well.

You can run a custom StartUp gcode but it must be written for Cura.  I'll remove the bug label and go ahead and close this.  I hope you understand.

closed 2024-04-08 by GregValiant

LowtechEnvy, 2024-04-08:
Yes, I did!

Thanks again for the quick response.