Ultimaker/Cura #18854: Manual tree support path making

T9Air, 2024-04-09:
### Is your feature request related to a problem?

Sometimes I want to have the tree support that hits a certain part of my model have a higher branch range than a different part of the model, so that supports are not printed on the model itself, such as when one part is harder to reach than the others, but the easier-to-reach spots need sturdier supports.

### Describe the solution you'd like

So my idea is this: 1. 2 main areas are chosen, the start point of the support, and the end point of the support. 2. You add additional points between the 2 points for the support to follow. 3. You choose if the other areas that need support are supported, or not (if yes, all other areas are supported based on standard settings.), 4. The slicer creates the best path based on the points for the support to follow. 5. If the other areas are supported, the slicer adds support for them.
I tried to show what I meant in the picture, but it might not come out so well.

### Describe alternatives you've considered

I have tried to use custom supports, but they do not seem as customizable.

### Affected users and/or printers

Anyone who has supports that they really want to customize.

### Additional information & file uploads

[image]

ThomasRahm, 2024-06-02:
I think the problem you are trying to solve can already be solved, but in a different way.
You can use "support blocker" to change the settings in an area of the model (instead of the default of blocking support). 
[image]

Currently one can't use regular support and tree support at the same time, but as you can adapt the branch angle, you can prevent the branches from moving.

T9Air, 2024-06-19:
Besides just using it to support models, with custom supports, you can create very interesting designs.

T9Air, 2025-03-30:
There seems to be something similar to what I want implemented with this extension: https://github.com/Slashee-the-Cow/CustomSupportsReborn

closed 2025-03-30 by T9Air