Ultimaker/Cura #18855: [5.4.0] Ghost Layers not printing anything

LostKatana, 2024-04-09:
### Cura Version

5.4.0

### Operating System

Windows 10 Pro

### Printer

Creality Ender 3 (modified)

### Reproduction steps

I designed something like a screwdriver for my sons toy in Fusion 360 and from there I sent the construction to Cura with the build in functionality.
The print Sequence is set to "One at a Time", what seems to cause the problem.

### Actual results

**One at a Time:**
Slicing the model results in 448 layers where the last 198 layers seem to do nothing but appear to be travels.
Starting from Layer 250 there is no visuals of any layer getting printed. In addition to that, after printing the last layer it just stays at the last position and the print head is not moving to home.

**All at Once:**
Slicing the model results in 226 layers and the last travel is

### Expected results

Printing "One at a Time" should not result in additional layers above the 250 layers for the model.

### Add your .zip and screenshots here ⬇️

[Toolio Screwdriver.zip](https://github.com/Ultimaker/Cura/files/14914167/Toolio.Screwdriver.zip)
[image]

GregValiant, 2024-04-09:
Thanks for the report.
I can duplicate this.  The displayed layers aren't right.  The Gcode is correct though so something seems to be going on with the display.
The gcode shows the layer_count of the tall model as 224 and the smaller model at 26.

Someone from the Cura team will take a look.

HellAholic, 2024-04-29:
Added a ticket for the investigation.
Internal reference: CURA-11881