Ultimaker/Cura #18858: Slicing Failed (5.7.0)

thurask, 2024-04-09:
### Cura Version

5.7.0

### Operating System

Windows 11

### Printer

Creality Ender 3 Pro

### Name abnormal settings

Still crashes with standard 0.2mm profile

### Describe model location

Rotation and moving the models around doesn't seem to change anything

### Describe your model

The model has printed successfully before, but attempting to slice 3x of the same clamp fails. No errors when loading and Mesh Tools fixes nothing.

### Add your .zip here ⬇️

[Clamp_locking_ring Polymaker ABS lw 0.4mm lh 0.2mm if 30% ext1 260C bed 95C 2024-04-09T160004.zip](https://github.com/Ultimaker/Cura/files/14923346/Clamp_locking_ring.Polymaker.ABS.lw.0.4mm.lh.0.2mm.if.30.ext1.260C.bed.95C.2024-04-09T160004.zip)

STL in question is Clamp_locking_ring.stl from here: https://www.printables.com/model/581285-clamp-for-tilting-forces-eibos-easdry-magnetic-ptf/files

HellAholic, 2024-04-09:
Hey @thurask ,
This looks similar to an issue we're currently investigating.
The log file shows a permission error "[error] The file %s could not be opened for writing."
The current quick workaround is to run Cura as administrator by doing "Right Click > Run as Administrator"

Thanks for sharing the model!

For internal Cura reference: CURA-11821

thurask, 2024-04-09:
Thanks for the workaround, that seems to have done the trick to get this to slice.