Ultimaker/Cura #18859: Fuzzy Skin never Outside Only

DSCustoms, 2024-04-09:
### Cura Version

5.7

### Operating System

Mac OS 12.7.4

### Printer

Flashforge

### Reproduction steps

Slicing with fuzzy skin

### Actual results

Fuzzy skin is selected outside only. It shows up on all inside surfaces, it shows up all over the place, sometimes it stripes around holes going inside and out, sometimes it just does this. The only time I've been able to use fuzzy skin is on small, featureless completely sealed objects. Disappointing!

### Expected results

I'd like the fuzzy skin to be applied to exterior surfaces only. 

### Add your .zip and screenshots here ⬇️

[image]
[FCPRO_lidsolid.3mf.zip](https://github.com/Ultimaker/Cura/files/14923973/FCPRO_lidsolid.3mf.zip)

GregValiant, 2024-04-09:
Thanks for the report.
I think what's going on is that Cura works by layers.  As the print progresses and "islands" develop the outside is all around each island.  So each layer is correct, but the effect is wrong.
The round feature in the back corner is a good example.  Down low it's attached to the wall so it is all "inside" and there is no fuzzy skin.  Then the top starts, is no longer attached to a wall so it is an island and it is all "outside" so it gets the fuzzy skin.

I'll leave the bug label but I have serious doubts that it is fixable.  Horizontal Hole Expansion suffered from something like this and a new setting came up to address it.  Maybe @smartavionics has a thought?

DSCustoms, 2024-04-09:
You also replied to my other comment with the window pane suggestion, here's a file with a single line thickness window pane that also loses its mind with fuzzy skin. I wish there was a way to actually designate the surfaces to be fuzzy, like a face selection. 

[image]
[image]
[image]
[demowindow.3mf.zip](https://github.com/Ultimaker/Cura/files/14925129/demowindow.3mf.zip)

DSCustoms, 2024-04-09:
Whenever I try the windowpanes if I don't make them seriously thick, this happens. Once you are making something with a 1.2 mm thick wall, it's no longer easy to remove. I tried giving one cut lines but those spots failed as well. The behaviour of switching walls from inside to outside every time it meets a window is the real pain. Why does it lose its mind and give those smooth areas on the outside walls once it flips on a window?

GregValiant, 2024-04-10:
Slide the top preview button down to layer one, then use the bottom button to slide the preview up.  It will show one layer at a time.  You can see how the "all the way around fuzziness" occur every time there is a break in the outer wall.
The window panes need to be 2 walls thick.  If they aren't, then they don't work to block the fuzzy skin from sneaking inside.  The thickness that's required might have something to do with the fuzzy settings.
I don't see a way to fix this.  A plain box will print fine but if there are gaps in a wall then everything becomes Outside Wall.
Fuzzy Skin on the Outside Only needs Fuzzy Logic.  "I know that looks like a hole but don't treat it like a hole" sort of thing.  I don't see that happening.

smartavionics, 2024-04-10:
Hi, yes, I have a cunning plan that could yield a solution. @DSCustoms , could you please provide the project file for the box you show in your first picture. I will take a look at this over the next couple of days.

GregValiant, 2024-04-10:
@DSCustoms If you hit the "Comment" button too fast the link you upload gets butchered into something that doesn't work.  In this case it turned into a string of text with no hyperlink.
[Project_Fuzzy Test.zip](https://github.com/Ultimaker/Cura/files/14931434/Project_Fuzzy.Test.zip)

DSCustoms, 2024-04-10:
Ok lets try again. 
[FCPRO_lidsolid.3mf.zip](https://github.com/Ultimaker/Cura/files/14931558/FCPRO_lidsolid.3mf.zip)

smartavionics, 2024-04-13:
Thanks for the 3mf. Unfortunately, my Cura doesn't recognize your printer so I just get the model with no settings. It doesn't matter though. I have a solution that works for this model but it won't work for all models. It's quite a tricky problem to detect the surfaces that are "inside" the model. One problem with this solution is that it may misbehave if either there are multiple models on the buildplate or if the model outline has concave regions. It may be that other solutions would cope better in that situation so I would be interested in trying more complicated models.

Anyway, Here's what I see...

[image]

I see you are using a Mac. Sadly, I can't build a Cura release for that system so I can only provide some code diffs that maybe the UM developers will pick up. The changes are pretty trivial...

```
diff --git a/src/FffPolygonGenerator.cpp b/src/FffPolygonGenerator.cpp
index 335516da2..aa25c1273 100644
--- a/src/FffPolygonGenerator.cpp
+++ b/src/FffPolygonGenerator.cpp
@@ -1268,17 +1268,20 @@ void FffPolygonGenerator::processFuzzyWalls(SliceMeshStorage& mesh)
     const coord_t avg_dist_between_points = mesh.settings.get<coord_t>("magic_fuzzy_skin_point_dist");
     const coord_t min_dist_between_points = avg_dist_between_points * 3 / 4; // hardcoded: the point distance may vary between 3/4 and 5/4 the supplied value
     const coord_t range_random_point_dist = avg_dist_between_points / 2;
+    const coord_t wall_line_width_0 = mesh.settings.get<coord_t>("wall_line_width_0");
+    const bool outside_only = mesh.settings.get<bool>("magic_fuzzy_skin_outside_only");
     unsigned int start_layer_nr = (mesh.settings.get<EPlatformAdhesion>("adhesion_type") == EPlatformAdhesion::BRIM)? 1 : 0; // don't make fuzzy skin on first layer if there's a brim
     for (unsigned int layer_nr = start_layer_nr; layer_nr < mesh.layers.size(); layer_nr++)
     {
         SliceLayer& layer = mesh.layers[layer_nr];
+        Polygons hull = layer.getOutlines(true).approxConvexHull(0);
         for (SliceLayerPart& part : layer.parts)
         {
             Polygons results;
             Polygons& skin = (mesh.settings.get<ESurfaceMode>("magic_mesh_surface_mode") == ESurfaceMode::SURFACE)? part.outline : part.insets[0];
             for (PolygonRef poly : skin)
             {
-                if (mesh.settings.get<bool>("magic_fuzzy_skin_outside_only") && poly.area() < 0)
+                if (outside_only && poly.area() < 0)
                 {
                     results.add(poly);
                     continue;
@@ -1288,8 +1291,21 @@ void FffPolygonGenerator::processFuzzyWalls(SliceMeshStorage& mesh)
 
                 int64_t dist_left_over = rand() % (min_dist_between_points / 2); // the distance to be traversed on the line before making the first new point
                 Point* p0 = &poly.back();
+                unsigned pindex = 0;
                 for (Point& p1 : poly)
                 { // 'a' is the (next) new point between p0 and p1
+                    if (outside_only)
+                    {
+                        // create a point from the vertex that should be outside of the hull if the wall is an outside wall
+                        Point outside = PolygonUtils::getBoundaryPointWithOffset(poly, pindex++, wall_line_width_0);
+                        if (hull.inside(outside))
+                        {
+                            // the point is inside the hull so assume the wall is an inside wall and so add it un-fuzzed
+                            result.add(p1);
+                            p0 = &p1;
+                            continue;
+                        }
+                    }
                     Point p0p1 = p1 - *p0;
                     int64_t p0p1_size = vSize(p0p1);
                     int64_t p0pa_dist = dist_left_over;

```

smartavionics, 2024-04-13:
https://github.com/Ultimaker/Cura/assets/585618/716ac06b-1d94-4e28-9a1c-465df983769c

smartavionics, 2024-04-13:
Just discovered an unrelated problem with fuzzy skin as it doesn't mix with bridging very well. It really should not fuzz the bridge wall lines. Not easy to fix, add it to the bug list!

[image]

GregValiant, 2024-04-13:
Now there is an unanticipated result.  As far as bugs go I believe it's gonna be pretty far down the list.

It won't really look that that.  It could touch the inside wall string and get stuck to it and might keep some of that shape but for the most part it will be straight as it follows the nozzle.

smartavionics, 2024-04-13:
So what I have done for my Cura is to not print the outer wall using bridge wall settings when it is fuzzed. That should make it thicker (depending on the settings) which should make it more likely to stick to the inner walls.

[image]

smartavionics, 2024-04-14:
BTW, here's a slight tweak to the code shown above. The part of the code that outputs the un-fuzzed line is better like this...
```
+                        if (hull.inside(outside))
+                        {
+                            // ensure that the wall starts on the previous vertex
+                            if (result.size() > 0 && result.back() != *p0)
+                            {
+                                result.add(*p0);
+                            }
+                            // output the original un-fuzzed wall
+                            result.add(p1);
+                            p0 = &p1;
+                            continue;
+                        }
```

MariMakes, 2024-05-01:
Hey @DSCustoms,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 
I encounter this as well sometimes and understand it can be annoying. 

Thanks for the helpful comments @smartavionics 
We discussed your suggestions with the team and are going to try to improve this. 
We've added a ticket to the backlog, for internal reference CURA-11887

Thanks for the report! 👍

DSCustoms, 2024-05-01:
> Hey @DSCustoms,
>
> Welcome to the UltiMaker Cura Github 🚀 Sorry, it took us a while to get back to you 😞 I encounter this as well sometimes and understand it can be annoying.
>
> Thanks for the helpful [...]

Appreciate the reply. I know this is far more complicated than I could ever fix myself, I just find I make lots of enclosure type prints and I’d love a way to use fuzzy skin on them, and this pops up every time. Being able to highlight skin layers somehow and have them print the fuzzy layer would be awesome, like bambu studio does with their multi color prints.

rburema, 2024-07-04:
Did some investigation & experimentation based on the findings here. See [the tentative result here](https://github.com/Ultimaker/Cura/pull/19319), and [for the backend, here](https://github.com/Ultimaker/CuraEngine/pull/2113).