Ultimaker/Cura #18860: Cura 5.7 | The settings for the last layer of spiralize outer contour. How to cancel the last layer?

Eazao, 2024-04-10:
### Cura Version

5.7

### Operating System

Windows11

### Printer

Eazao Ceramic 3D printer

### Reproduction steps

We recently upgraded Cura to the latest version, Cura 5.7, and turned on "spiralize outer contour" in the settings, but did not turn on "enable ironing". We slice and print the cylindrical model. At the end of the cylindrical model, the nozzle will maintain the final Z value and print another circle, as shown in the figure below. How to turn off this feature? There is no such problem in Cura 4.13.

### Actual results

We recently upgraded Cura to the latest version, Cura 5.7, and turned on "spiralize outer contour" in the settings, but did not turn on "enable ironing". We slice and print the cylindrical model. At the end of the cylindrical model, the nozzle will maintain the final Z value and print another circle, as shown in the figure below. How to turn off this feature? There is no such problem in Cura 4.13.

### Expected results

We recently upgraded Cura to the latest version, Cura 5.7, and turned on "spiralize outer contour" in the settings, but did not turn on "enable ironing". We slice and print the cylindrical model. At the end of the cylindrical model, the nozzle will maintain the final Z value and print another circle, as shown in the figure below. How to turn off this feature? There is no such problem in Cura 4.13.

### Add your .zip and screenshots here ⬇️

[image]
[UCP_cylinder-50-100mm.zip](https://github.com/Ultimaker/Cura/files/14926078/UCP_cylinder-50-100mm.zip)

HellAholic, 2024-04-10:
Try Surface Mode > "Surface" in the special modes (Should be above the Spiralize Outer Contour option).
Since your printer is not included in Cura this is just a suggestion, I cannot be sure what settings you have enabled or what sort of behavior would happen with changing settings.
[image]

There is a major change moving from 4.X to 5.X so changes in behavior is to be expected. There are forks of Cura based on the 4.13 version that the community members maintain and update.
This is a discussion type for help needed.

Also note for future tickets: Please don't copy paste the same text in all fields, leaving them empty is also an option if you don't have specific information for that field.

Eazao, 2024-04-10:
> Try Surface Mode > "Surface" in the special modes (Should be above the Spiralize Outer Contour option). Since your printer is not included in Cura this is just a suggestion, I cannot be sure what [...]

Thank you for your suggestion.

We have previously tested "Surface" mode, as well as turning it on at the same time, but the problem still exists. You can view the attachment, which is our project file.

Thanks for the reminder, we will pay attention to these in the future, due to github settings, these questions must be answered, otherwise they cannot be submitted.

[UCP_cylinder-50-100mm.zip](https://github.com/Ultimaker/Cura/files/14931261/UCP_cylinder-50-100mm.zip)

HellAholic, 2024-04-10:
Hey @GregValiant , do you have any experience with this issue and how to resolve it?

GregValiant, 2024-04-11:
Slashee_the_Cow and I went over this yesterday.  We had a "script" competition.

Spiralize creates a "ramp" all the way up the model.  If the nozzle simply stops ramping upwards then the top of the model isn't planar to the build surface and it has a step where the extrusion ended and the nozzle pulled away.
The current implementation eliminates both of those problems by doing a second loop around the model on the top layer.  As the second loop progresses the layer height diminishes (because of the helix) and the flow is compensated so the line width remains correct.

There are three ways to accomplish what the user wants:

- Revert to 4.13.1 which makes no attempt to flatten the top layer.
- Post process a 5.x Cura file to get rid of just the non-spiral part of the top layer.
- Post process to remove the entire top layer.  

Slashee posted her script on the UM Forum.  It's of the "quick and dirty" variety.
My attempt is more ambitious and ambidextrous as it has options for points 2 and 3 above as well as accounting for the final retraction in the file (if the user happens to have retractions enabled).  It was an interesting exercise.  I really don't see a lot of use for this.

Eazao, 2024-04-12:
> Slashee_the_Cow and I went over this yesterday. We had a "script" competition.
>
> Spiralize creates a "ramp" all the way up the model. If the nozzle simply stops ramping upwards then the top of the [...]

thank you for your reply.

If we still want to use Cura 5.7, how can we adjust the extrusion flow of the last layer?

Frankly speaking, we also think that the current "Spiralize" solution of Cura 5.7 has its advantages, because there is no step in the end. But you can see the picture below. The current method has a relatively large extrusion volume and will interfere with the already printed ones. How to further reduce the extrusion flow when printing the last layer?
[image]

Eazao, 2024-04-12:
In addition, we still have 2 questions regarding spiralization:

1. This is the ceramic cup we printed. How to achieve the finishing effect as shown in the picture in Cura5.7? We cannot adjust the final traffic. Picture 1 is the result of using the blade after printing on Cura 4.13.

2. For spiralization, picture 2, in the initial layer, in the very beginning part, the bottom layer will bulge. How to adjust this part to make it even and consistent with the others?

[image]

GregValiant, 2024-04-12:
The "variable flow" that 5.7 uses to start the ramp and to finish the ramp is the fix for what you are showing in the last 4.13 image of your post above.  That ridge at the bottom was the problem.

The bottom layers were all of "layer height" and then the spiral would start.  But at the start of the spiral, the initial Z gap above the previous layer was only a fraction of a layer height with the volume of filament calculated as if the gap was a full layer height.  That resulted in over-extrusion at the start of that loop around the part and the characteristic bulge.  The over-extrusion would lessen as the gap between the previous layer and the new layer incrementally increased until by the end of the loop the new layer was in fact 1 layer height above the previous layer.  From that point up, each point on the new layer is 1 layer height above the same point on the previous layer.
The same fix is applied to the top to avoid a bulge as the print finished.
This is the same method used on something like a coil spring where flats are created on the top and bottom so there is no "step" when the coils stop.
[image]

So it appears that you like the bottom part of the fix, but not the top part.

Each spiral layer is 1 loop around the part.  The top layer is two loops around the part.
You can delete the entire top layer or you can delete just the "flattening" code that is the second loop in the top layer.
It would be possible to stick in an M221 line at either the start of the top layer, or at the start of the last loop (where the spiral ends).  I don't see that as a fix because the second loop is planar parallel to the base plate and so the helix of the first loop creates the "decreasing Z gap" problem that is the opposite of what happens at the start of the spiral.

The end of the spiral can be found mid-layer because the Z movement ends.  Deleting the second loop of the top layer should allow you to have the 5.7.0 start of the spiral (that has variable flow) and the non-variable top of the spiral like you see in 4.13.
You can't really make a flow adjustment because the gap between "top layer -1" and top layer keeps getting smaller.  I would think that simply getting rid of the second loop of the top layer would be sufficient.  It might need a wipe there, but it should work.

GregValiant, 2024-04-12:
The original fix for the "bulge" problem came from @smartavionics and his fork of Cura (which is based on Cura 4).  Maybe he can address this "leave the bottom fix but turn off the top fix" which is my impression of what you are looking for.

smartavionics, 2024-04-12:
Hello @Eazao , have you tried my Cura variant? As Greg says, it's based on Cura 4 but it does have a few tweaks compared to UM Cura. You can find my releases at https://github.com/smartavionics/Cura/releases

Eazao, 2024-04-12:
> The "variable flow" that 5.7 uses to start the ramp and to finish the ramp is the fix for what you are showing in the last 4.13 image of your post above. That ridge at the bottom was the problem.
> 
> The bottom layers were all of "layer height" and then the spiral would start. But at the start of the spiral, the initial Z gap above the previous layer was only a fraction of a layer height with the volume of filament calculated as if the gap was a full layer height. That resulted in over-extrusion at the start of that loop around the part and the characteristic bulge. The over-extrusion would lessen as the gap between the previous layer and the new layer incrementally increased until by the end of the loop the new layer was in fact 1 layer height above the previous layer. From that point up, each point on the new layer is 1 layer height above the same point on the previous layer. The same fix is applied to the top to avoid a bulge as the print finished. This is the same method used on something like a coil spring where flats are created on the top and bottom so there is no "step" when the coils stop. ![spring](https://private-user-images.githubusercontent.com/64202104/321957160-28cfa405-3f14-4ef2-804d-c5894f96a076.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTI5MzIyOTcsIm5iZiI6MTcxMjkzMTk5NywicGF0aCI6Ii82NDIwMjEwNC8zMjE5NTcxNjAtMjhjZmE0MDUtM2YxNC00ZWYyLTgwNGQtYzU4OTRmOTZhMDc2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MTIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDEyVDE0MjYzN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWU5ZTUyZjBmNWFiOTBjN2Y3ZDhjYzYwZDVhMzcxMWRkMzQ2YzNlMjQyMWI5NGM0Nzc3MTdiYTI1NTUzZGM3NDAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0._4RDBAs8fCTnibNlGiHJCpxxnRMOWdrBSbW5qHMfGYI)
> 
> So it appears that you like the bottom part of the fix, but not the top part.
> 
> Each spiral layer is 1 loop around the part. The top layer is two loops around the part. You can delete the entire top layer or you can delete just the "flattening" code that is the second loop in the top layer. It would be possible to stick in an M221 line at either the start of the top layer, or at the start of the last loop (where the spiral ends). I don't see that as a fix because the second loop is planar parallel to the base plate and so the helix of the first loop creates the "decreasing Z gap" problem that is the opposite of what happens at the start of the spiral.
> 
> The end of the spiral can be found mid-layer because the Z movement ends. Deleting the second loop of the top layer should allow you to have the 5.7.0 start of the spiral (that has variable flow) and the non-variable top of the spiral like you see in 4.13. You can't really make a flow adjustment because the gap between "top layer 1st loop" and "top layer 2nd loop" keeps getting smaller. I would think that simply getting rid of the second loop of the top layer would be sufficient. It might need a wipe there, but it should work.

Thank you very much for your explanation, we will try further.

Eazao, 2024-04-13:
Hi, [smartavionics](https://github.com/smartavionics), [GregValiant](https://github.com/GregValiant),

Thanks for your suggestions. We recently tested Cura 4.20.18, turned on spiral printing, and the effect of the last layer was good.

In addition, the initial part of the layer has also been improved.

https://github.com/smartavionics/Cura/releases
[image]
[image]

Eazao, 2024-04-19:
Hi [smartavionics](https://github.com/smartavionics),
Hello, the red line in the picture is the scratch caused by the empty path of cura 4.20.18 version. How to close it?
[image]
The picture below is the path of Cura 5.7, without this scratch.
[image]

smartavionics, 2024-04-19:
Hi @Eazao , for issues related to my Cura variant, please open a new issue at https://github.com/smartavionics/Cura/issues rather than post here so not to distract the UM devs. Please provide a project file for the above print so I can investigate. Thanks.

Eazao, 2024-04-19:
> Hi @Eazao , for issues related to my Cura variant, please open a new issue at https://github.com/smartavionics/Cura/issues rather than post here so not to distract the UM devs. Please provide a project file for the above print so I can investigate. Thanks.

Thanks for your suggestion, we have submitted: https://github.com/smartavionics/Cura/issues/249