Ultimaker/Cura #18868: Bridging Settings When Utilized for Sparse Infill Applies Beyond 1 Layer (Or 3 with Multiple)

mzhu1113, 2024-04-10:
### Cura Version

5.7.0

### Operating System

Win 11

### Printer

NA

### Reproduction steps

1. Utilizing Bridge Sparse Infill Density
2. **Every** top skin layers get bridging settings applied to it for the immediate skin layer after infill. This only applies to where the sparse infill condition is true.

This issue is present even if Bridge has Multiple Layers is off so these bridging settings are applying even after Layer 3 of that top skin.

This is a problem as this causes the top skin, if bridging settings apply, to run in only one direction with rectlinear pattern, with less density, and more/less flow which is an undesirable behavior.

I cannot provide project files as the current job is under NDA but this issue is present regardless of the model so long the bridge settings applies for the sparse infill.
I attached a screenshot of the settings at the very least so if there's something wrong with the chosen settings let me know and I can correct/be enlightened/etc.

This issue is also present in 5.6.0.

I only noticed this issue now because I have Sparse Infill set to 10% and Infill set to 20%. Yet, the bridging settings are somehow applying. This value has always been somewhat dubious in its behavior but setting it at 10% was mostly a safe bet until 5.7.0 where this value is seemingly ignored despite the infill % being firmly outside the range of it being "sparse".

Thanks as always!

### Actual results

The top skin layers have bridging settings applied.

### Expected results

Only the 1 or 3 layers would have bridging settings but not layers 4+

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2024-04-10:
Thanks for the report.
I can duplicate this.  It appears that it only works if "Bridge Sparse Infill Max Density" is set < "Infill Density / 2".  

With the Infill Density at 50%:
- If Sparse Density is 24.9% there is no bridge over the infill.
- With Sparse Density at 25% there is bridge over the infill.

The Cura team will take a look.

HellAholic, 2024-04-10:
Hey @mzhu1113 ,
There are as you mentioned couple of factors in play.

Example: with infill pattern of Grid -> infill density of 60% -> Bridge Skin Support Threshold of 45 generates bridging

[image]
[image]

, but if you change your infill pattern of Lines then the same settings will result in no bridging.

[image]
[image]

HellAholic, 2024-04-10:
It's the other way around :P
Grid has half the sparse density of lines, since you're crossing the lines at 90 degree-ish angle, there are half as many lines in either of the two given direction.

mzhu1113, 2024-04-10:
In response to both of you, thanks for the replies!

It is definitely confusing. I assumed it was related to the Infill line Distance values which sort of basically helps give a reference but the behavior changes when the Bridge Support Threshold is in play. Do either of you happen to know what the full formula is to flag a skin to be using bridging settings? Right now the only thing that seems to fix the problem I'm having (Layers 4+ retaining bridging settings) is to change the Bridge Support Threshold to 95% instead of 99%. Honestly... I've long forgotten why I have it at 99%.

HellAholic, 2024-04-10:
In terms of actual calculation formula, I would have to poke the developers.
But the number of Top layers doesn't affect the bridging over infill. 
It calculates how much of area of the layer is being supported by the infill, if that value is higher than the Support Threshold, then it is all good, if it is lower than that value, then it will follow the bridging settings.

With regards to why you had it at 99%, I would guess as to prevent pillowing. 
I remember a few issues with bridging (over)/(on top of) infill that were resolved in 5.6 or 5.7

I'll see if I can get you the formula :P but no promises.

mzhu1113, 2024-04-10:
Haha yeah I remember I had this to help with pillowing and give the smoothest possible top layers without extending print times.

If you manage to get that formula, I would be super grateful! 
The number of top layers doesn't affect it but the bridging settings are affecting all those layers that have been flagged to use the bridging settings so basically every layer after the first top layer is also acting as if it is still bridging the initial layer.
I'm kind of wondering if the 99% threshold + sparse density settings is somehow factoring in the bridge's density too (92.5%) and thus causing a cascading effect making every top skin layer rectilinear and in one direction. I had to throw away a 2 day TPU part today after seeing gaps in the top layer and thus why we're here today haha.

HellAholic, 2024-04-12:
Hey @mzhu1113, I couldn't find/get a formula that would be clear enough. There were few pointers to look at the fdmprinter.def.json as most printer definitions inherit from that.
In addition, I forgot about the plugin by Ghostkeeper, the Settings Guide, https://github.com/Ghostkeeper/SettingsGuide
That one has a more elaborate explanation of the bridging settings (and few other ones).