Ultimaker/Cura #18870: Moving multiple objects on the build plate

FreyJuan, 2024-04-10:
### Cura Version

5.7

### Operating System

Windows 10 GeForce 1060(possibly)

### Printer

Anchor 5

### Name abnormal settings

It sliced at first.

### Describe your model

Models seem good, they were very crowded together and I was moving them apart.

### Add your .zip here ⬇️

[RHD_5mm_Post_10_Top_Back.zip](https://github.com/Ultimaker/Cura/files/14937213/RHD_5mm_Post_10_Top_Back.zip)

GregValiant, 2024-04-11:
Thanks for the report.
I didn't have any trouble moving things around and slicing.

There have been some reports of this sort of behavior and it's being looked into.
Try right clicking on the Cura icon on the desktop and select "Run as Administrator" and see if there is a difference.
What language do you use on the computer?  There have been reports of non-latin based languages having an effect.

I'll mark this as a duplicate and the Cura team will take a look.

HellAholic, 2024-04-11:
Other things that you can try out:
- Disable auto slice since the intersection of many models can create shapes/paths that result in the fail (while your moving the models)
- Instead of moving the models one by one, right click and select arrange all models in grid