Ultimaker/Cura #18873: Item won't slice

LDDSurfer, 2024-04-11:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

ELEGOO Neptune 3 Pro

### Name abnormal settings

No major settings change

### Describe model location

45* rotated left and slightly angled down

### Describe your model

Model was just built, not sure if watertight, did use mesh fix

### Add your .zip here ⬇️

[New Compressed (zipped) Folder.zip](https://github.com/Ultimaker/Cura/files/14941285/New.Compressed.zipped.Folder.zip)

LDDSurfer, 2024-04-11:
10% is proper scale size

HellAholic, 2024-04-11:
Please save the model as a project file as it includes your printer settings and custom profiles and other important details that could cause the issue. Including the cura.log also helps, to retrieve go to Help > Show Configuration folder > zip the cura.log and drop that one in as well with the project file.