Ultimaker/Cura #18874: I cannot slice this particular model on the printer. I have sliced it before weeks ago, so it shouldn't be a problem

JonathanJensenn, 2024-04-11:
### Cura Version

5.7.0

### Operating System

Windows 11 PRO

### Printer

Ultimaker S5

### Name abnormal settings

No, i have used these settings succesfully before. No it doesnt slice with default profiles.

### Describe model location

No

### Describe your model

Yes i have sliced it succesfully. Yes watertight. No mesh fix.

### Add your .zip here ⬇️

[image]

HellAholic, 2024-04-11:
Without the project file I can't really offer any help.
- Try running cura in administrator mode and see if that resolves the issue, if not please provide a project file.

JonathanJensenn, 2024-04-12:
Hi,

Running Cura in administrator mode doesn't work for me.

The file is too large to be attached here, so i've uploaded it to WeTransfer. Download link below.

https://we.tl/t-LwjNx6VbGS
UMS7_RM logo crockery 3 TEST.3mf<https://we.tl/t-LwjNx6VbGS>
1 file sent via WeTransfer, the simplest way to send your files around the world
we.tl

Thanks!

[https://img.mysignature.io/p/7/5/d/75d36ae8-4da2-5ab4-aac0-d25ac2fd29a2.png?time=1651140837]<https://linktr.ee/logumklosterlys>

Jonathan Jensen

Trainee @ Løgumkloster Lys A/S

Check us out HERE!<https://www.logumklosterlys.com/>

***@***.***,8.94825,15z/data=!4m5!3m4!1s0x0:0x1e08c159a70f39ea!8m2!3d55.0517555!4d8.9482529>

[https://img.mysignature.io/s/v3/c/8/0/c802bf5f-50cd-58f1-b225-95aed4c481f3.png]<https://www.instagram.com/lkl.candle.lab>

[https://img.mysignature.io/s/v3/d/f/e/dfede595-3e57-516c-a753-6df9185e85f4.png]<https://www.linkedin.com/company/logumklosterlys>

HellAholic, 2024-04-12:
[UMS7_RM logo crockery 3 TEST.zip](https://github.com/Ultimaker/Cura/files/14954892/UMS7_RM.logo.crockery.3.TEST.zip)
Can reproduce the issue. Posting the model here since the we transfer links expire (and less trusted).

HellAholic, 2024-04-12:
Quick update: The issue relates to the interaction with the size of  the data in the layers and gradual flow plugin. If Gradual flow is disabled for both extruders, the model slices.
Side note: Model has some issues that can be repaired but the issue is not caused directly by those.

For internal reference of Cura team: CURA-11839