Ultimaker/Cura #18875: skipping a layer on thin lines?

illusionmanager, 2024-04-11:
### Cura Version

5.7.1 beta

### Operating System

windows 10

### Printer

snapmaker with dual head

### Reproduction steps

- install snapmaker plugin https://marketplace.ultimaker.com/app/cura/plugins/Snapmaker/SnapmakerPlugin
- load base and assign the first extruder with 0.4mm nozzle to it (just a disk in blue)
- load moonphases and assign the second extruder to it (with 0.2mm nozzle)
- select both model and merge them
- set layer height to 14 and initial layer height to 16

### Actual results

in this image, layer 21 is the top of the blue base; layer 22 is the first layer of the moonphases with the rings around each moon, printed correctly; layer 23 is the second layer of the moonphases and somehow cura skips in this layer the rings but not the moons, layer 24 prints again correctly.

### Expected results

it should have printed the rings in all three layers on top of the blue. Initially I modeled the rings and moons separately with this result, so I decided to use the boolean operator in 3dsmax to make the ring and the moon into one object, that is what is loaded into this project. The rings and the moons are exactly the same height.
The rings have a width of 0.405mm (in 3dsmax) and shouldn't be too thin for a 0.2mm nozzle to print correctly in all layers.

### Add your .zip and screenshots here ⬇️

[image]
[moonphases.zip](https://github.com/Ultimaker/Cura/files/14944387/moonphases.zip)

GregValiant, 2024-04-11:
Thanks for the report.
The Moons aren't watertight.  If there are open seams or reversed surfaces in the mesh it can fool a slicer and you end up with missed layers.
After repairing the model with MS 3D Builder all the layers show up.
Here is a report from an online STL repair site.
-> Reading file and indexing vertices
-> Analysed your file:
--> 0 Naked edges (?)
--> 0 Planar holes (?)
--> 0 Non-planar holes (?)
--> **3 Non-manifold edges** (?)
--> **4 Inverted faces** (?)
--> **8888 Degenerate faces** (?)
--> 0 Duplicate faces (?)
--> 0 Disjoint shells (?)

Once it is repaired it slices just fine.
[image]

This is a model problem so I'll remove the bug label and go ahead and close this.  I hope you understand.

illusionmanager, 2024-04-11:
hmm, i already figured that the boolean operation might have created a bad stl, so I fixed one of them and the problem didn't change. Also the problem was already there when I used separate objects for the rings and the moons, leaving the "boolean" work to cura. I *just* found however a workaround that I was going ta add to the bug report. Instead of exporting all the rings and all the moons as a single group in 3dsmax, I combined each ring with a moon (still using boolean) and exported those 16 objects. Loading the 16 objects into cura worked fine.

illusionmanager, 2024-04-11:
@GregValiant i just wanted to thank you for the pointer to ms 3d builder. I never used that program, but it fixed the errors just fine. Even for my snapmaker printer. Thanks!

GregValiant, 2024-04-11:
MS 3D Builder is pretty good.  Not intuitive mind you, but competent.  I also use it for altering STL's when I don't have the CAD file.
From what I've seen around here probably 75% of the bug reports involve bad models.  Another 15% are users misunderstanding the software.  Of the 10% that are left there is sometimes a workaround.  The Cura team takes them all seriously.  Nobody likes buggy software and 5.4 - 5.6 were poor with all the slicing errors.  5.7.0 stable is a huge step forward as many of the slicing errors were addressed.  New settings like the "interleaved" prime tower will have growing pains but that's a normal progression.

illusionmanager, 2024-04-11:
I just loaded the model into ms 3d builder. it offered to repair it, and I saved it. Only problem is that apparently it changed the z coordinates, so the model didn't merge correctly anymore in cura.

If 75% of the problems are from bad models, it might be worth investigating into automatically repairing bad models in cura itself. :-D

GregValiant, 2024-04-11:
The repair utility "Mesh Tools" has some basic fix up capability.  Prusa made some sort of deal with AutoDesk and uses NetFabb as an external app to repair models.
Very few of my own models (designed in AutoCad) ever have a problem.  Some of the downloaded models (Thingiverse and Printables) are seriously messed up.  Apps like Blender than weren't really designed to produce printable solid models are pretty bad as well.  If the user is aware that there can be model problems that affect the slice then going to an outside app isn't really a problem.  Outside apps can be dedicated to fixing while the slicers can concentrate on slicing.
I own 12 hammers.  Pounding in a tack with a 24oz ball peen hammer doesn't work well and driving a #20 barn nail with a welding hammer doesn't work so well either.  Ya gotta use the right tool for the job.