Ultimaker/Cura #18876: Printer doesn't clean nozzle when changed.

GarmTerje, 2024-04-11:
### Cura Version

5.7.0

### Operating System

Windows 10 version 22H2

### Printer

Ultimaker S5 w/material station and air manager

### Reproduction steps

Printed using white filament and 0.8 AA nozzle.
Changed to 0.4 AA nozzle that previously was used printing black filament.
When started a new job still printing white filament, the printer did not clean the changed nozzle.

### Actual results

The lack of cleaning the changed nozzle resulted in a couple of black layers before the output from the changed nozzle became white.

### Expected results

The printer detects changing the nozzle. Hence a cleaning flag should be set in the software before staring a new print-job with same filament type.

### Add your .zip and screenshots here ⬇️

This is an internal printer software issue. No files or screenshots needed.

GregValiant, 2024-04-11:
Thanks for the report.
Have you mentioned this over on the [UM Forum](https://community.ultimaker.com/)?  I figure this sort of thing must have come up previously.

HellAholic, 2024-04-11:
Hey @GarmTerje , this is not a Cura issue. While we are part of the same company, different teams work on different parts of the offering as UltiMaker. As Greg suggested, please follow up in the UltiMaker Forum or through the support channels for the printer.

closed 2024-04-11 by HellAholic