Ultimaker/Cura #18877: brim tower printing in thin air with dual head

illusionmanager, 2024-04-11:
### Cura Version

5.6.0

### Operating System

windows 10

### Printer

snapmaker with dual head but tried preview with ultimaker 3 as well

### Reproduction steps

load a model that has some layers just in a single color.

### Actual results

after it printed a few layers correctly with extruder 1 only, it reached a point where it needed to switch to extruder 2. When printing the prime tower, it printed the material for extruder 2 in thin air. This can be seen in the preview attached. The tower is only printed for the color in use, instead of the colors that are going to be used. The outer tower is in blue, but the inner tower in yellow isn't printed during layers without yellow. It should have been.

### Expected results

if a color isn't used for the current layer, but will be used in future layers, the part of the tower currently skipped should be printed (in the color of the current layer to prevent useless extruder switches)

### Add your .zip and screenshots here ⬇️

[image]
[moonstest-base.zip](https://github.com/Ultimaker/Cura/files/14946429/moonstest-base.zip)

GregValiant, 2024-04-11:
Thanks for the report.
I can duplicate this with Cura 5.6.0 but the problem appears to have been fixed in 5.7.0 (which is much better BTW).

In regards to 5.7.0 - I do see another problem though.  When I switch the Prime Tower back and forth between "Normal" and "Interleaved" Extruder 2 is always "Interleaved" except for the top layer.
[image]

Something wonky is going on there.  The Cura team will take a look.  This might get closed in favor of a report about the 5.7.0 behavior.  I hope you understand.

illusionmanager, 2024-04-11:
I just tested it again with 5.7.0 beta-1 and it prints a different prime tower that doesn't have this issue.

illusionmanager, 2024-04-11:
sorry @GregValiant I was closing this report (due to being fixed in 5.7.0 beta) while you were doing your research. For me it currently isn't a problem anymore, but if you want to reopen it so the cura team can have a look at the problem you discovered. go ahead.

closed 2024-04-11 by illusionmanager

illusionmanager, 2024-04-11:
interesting enough I just installed the 5.7.0 stable version but it hasn't an option to select the prime tower type. at least that is what I thought originally. However searching for it revealed the option. Apparently it was hidden. I wonder if it is a good idea to skip the switching of the extruder when it isn't used in a certain layer. Like printing those intermediate layers all in blue. This prevents extra heating up the unneeded extruder, and prevents extra material ouzing out of it onto the model.

GregValiant, 2024-04-11:
There are a lot of ways to set up for dual extrusions.  The "Mixing Hot Ends" have become more popular and they require a certain setup.  Conventional dual extruder machines require a different setup.  IDEX machines can have even more considerations.  Tweaking the Extruder StartUp and Extruder Ending Gcodes can help.  They are often ignored by dual extruder printer manufacturers.

If the prime tower doesn't continue up then you will have "printing over air" issues.  Since the areas of the prime tower are reserved for each extruder, each area has to continue to build up.

Another thing to remember is that UM does much of the heavy lifting in regards to writing Cura.  Necessarily that means it will always be slanted towards their printers.  They provide it for free to the rest of us and so if I were to add a second extruder to my Creality printer I'd probably need to find some workarounds.  Your SnapMaker is in the same situation.  Stock-out -of-the-box Cura may not be "just right" but Cura is highly customizable and a way around can usually be found.

All that being said, if the materials you are using have proper StandBy temperatures then for larger models there isn't a problem as the unused hot end has time to cool down after each tool change.  For small models, there just isn't time for cool down so yes, the unused nozzle can still be above the melt point of the material and ooze across the model.
As with any small model, sometimes printing two of them can help as each individual layer will take longer and the nozzle can cool down to stand-by temperature.

There is lots to think about when configuring a dual extruder machine.