Ultimaker/Cura #18878: cura crashes when in the material settings and slicing at the same time

iceman2511, 2024-04-11:
### Cura Version

5.7.0

### Operating System

windows 11

### Printer

modified ender 3 with modified marlin

### Reproduction steps

import model.
go to settings, material settings, then go to print settings
slice model

### Actual results

cura crashes 1-2 seconds after pressing slice

### Expected results

model should slice without crashing

### Add your .zip and screenshots here ⬇️

[front_center.zip](https://github.com/Ultimaker/Cura/files/14948786/front_center.zip)
[cura.txt](https://github.com/Ultimaker/Cura/files/14948902/cura.txt)
[image]
this is when you hit slice

HellAholic, 2024-04-11:
You have issues in your custom profile/settings
[image]
Not sure what plugins you are using, but they could be the source of the crash so check with plugins disabled.
You've set a raft base line width of "0.0" which in the settings shows an error. 
Also with a list of setting changes this long I would recommend double checking things and saving it as a new custom profile.
[image]

iceman2511, 2024-04-11:
no clue how, but i fixed it, but ill firstly answer the questions.
1. with all plugins disabled, it still crashed.
2. that was when i was adjusting one of my material profiles and didnt finish it when i uploaded the file, the crashes still occurred when i fixed that.
3. also saving it to a new profile didnt work.

but then just after this i did a few things to cura and it magically fixed, heres what i did:

firstly i didn't want the settings to be saved to a profile, because i have all my settings for separate filaments in a plugin called material, so i went to my backups and reverted to my latest backup.

then since i recently installed the 5.7.0 release, updating from the beta, i went to delete the old 5.7.0 beta installation in file explorer.

i decided to open up the file again, enable all plugins and restart, and resize the model to how i like it, and decided to test it again to see what i did fixed it, and it did fix it, it didn't crash, and fully sliced.

Update:
ok, this is weird, I'm writing this around 5 mins after i wrote the above, I've now found the underlying issue here, and its extremely weird, basically having a fresh backup of your settings, and not changing any settings afterwards fixes the crashing, but right when you change any setting only within the materials print settings, and then straight after you press slice, then cura crashes.

HellAholic, 2024-04-11:
It takes a couple of "insert time unit" for the newer versions of the plugins to be in line with the latest Cura version.
I am glad that you managed to figure out the issue on your device. In each version of Cura we add new and shiny new settings that allow you more access to modify and customize, side effect of that is sometimes you get issues like the one you're experiencing. 
Getting back to the main issue, if you don't mind, go to Help > Show configuration Folder, zip that entire folder (.../cura/5.7) and post it here. The configuration folder is the best way to reproduce the type of issue you're experiencing since it allows us to just copy it over and run Cura in almost identical conditions as what it would be on your device.

iceman2511, 2024-04-11:
im not sure if its updated for you, but ive edited my comment a few times updating what ive found.

heres the zip file for the config folder:
[5.7.zip](https://github.com/Ultimaker/Cura/files/14950139/5.7.zip)

HellAholic, 2024-04-12:
Went through the logs. With the addition of your comments and explanation it's a weird interaction between the different plugins and your input. There are no clear "Error" logs for the crash, but a lot of warnings for settings without value.
I followed your explanation step by step but could not reproduce the crash. changes settings in material > print setting > Changed settings and pressed slice immediately > no crash. Reverted the change and sliced again > also no crash.

If you found issues while interacting with the plugins I would re-offer my previous recommendation to either wait for new version of the plugins or alternatively if you feel like it, help the plugin developers by reporting the issues on their repository.

Cheers!

closed 2024-04-12 by HellAholic

iceman2511, 2024-04-13:
very strange, I did also test this with all my extensions disabled, and i could still reproduce this issue without any problems.

I'm now thinking that it could maybe have to do something with my GPU, because I've also had this issue with Minecraft of all things, where I turn on a shader, and after 20 mins of use, the game unexpectedly crashes, but when I test it with my integrated graphics, it doesn't crash at all, maybe this issue is being carried over by Minecraft, does cura run on openGL to render? or does it just use the usual directX rendering?

iceman2511, 2024-04-13:
after a quick test using my integrated graphics, I can now confirm that the GPU is the issue, so now this problem is most likely NVidias problem, since I'm also experiencing this issue on Minecraft.

I did all the basic troubleshooting like restart system, update GPU and CPU drivers to latest version, still nothing, I also followed some more specific guides online who could solve this issue, but still didn't work, my only theory is that its my specific system configuration which is causing these things to happen to my GPU, but that's all I have to go on.

if you want to look deeper into the Minecraft crashes, search (Minecraft exit code "-1073740791"), including the speech marks, but not the brackets.