Ultimaker/Cura #18879: Cura writing incorrect layer heights to output file? Viewed in FlashPrint.

DSCustoms, 2024-04-11:
### Cura Version

5.7

### Operating System

Mac OS Monterey 12.7.4

### Printer

Flashforge Creator Pro

### Reproduction steps

I've been having problems with my second extruder. Everything works from the extruder, its not clogged or jammed, but it would randomly fail to extrude layers and I couldn't figure out why. I started opening the output x3g files in flashprint and noticed that layers that are supposed to be .2mm (from a .4 nozzle) are sometimes showing up as .4 thickness. No reason at all for them to do that, they also skip walls and change two walls to one thicker one. the .4 nozzle can't spit out a .4 thick .8 wide wall no matter what it wants to.  (arms and buttons dual ext)

The .4 nozzle extruder will print walls just fine. It'll extrude supports and flows great when loading. At random layers when it needs to print a bridge or top surface it will just fail to extrude, killing prints. 

I'm also including a second file (arms right only) which is only on one active extruder, which has a .6 mm nozzle. That one never fails to print but has quality issues occasionally on a few layers. You can see the occasional .4mm layers stuffed in there with the intended .2 layers. 

Cura always shows .2mm layers in the thickness display. 

 Is this a visualization problem when the x3g is input into flashprint to view? 

### Actual results

Failing prints and strange output. 

### Expected results

Hopefully a good print. 

### Add your .zip and screenshots here ⬇️

[image]
[image]
[image]
[image]

[ARMS_RIGHT_ONLY.3mf.zip](https://github.com/Ultimaker/Cura/files/14949015/ARMS_RIGHT_ONLY.3mf.zip)
[ArmsandButtons_DUAL_EXT.3mf.zip](https://github.com/Ultimaker/Cura/files/14949016/ArmsandButtons_DUAL_EXT.3mf.zip)

HellAholic, 2024-04-11:
Your printer definition is not included in Cura "Flashforge Creator Pro", if you're using a forked version then I would suggest posting issues there as I have no way of checking the settings or reproducing your issues. This is simply because I don't have access to that printer profile.
I'll change this to a discussion type.

GregValiant, 2024-04-11:
There is a Cura[ tutorial here](https://community.ultimaker.com/topic/34903-tutorial-on-how-to-use-cura-with-flashforge-creator-pro/).  It looks like it's about 4 years old.  Have you followed something like that?

DSCustoms, 2024-04-11:
> There is a Cura[ tutorial here](https://community.ultimaker.com/topic/34903-tutorial-on-how-to-use-cura-with-flashforge-creator-pro/). It looks like it's about 4 years old. Have you followed something like that?

I’ve been using cura on the printer for a long time but I didn’t use dual extruders from it often. Mostly just had the one disabled and ran the .6 nozzle. Trying to get detailed prints led me to investigate this more. I don’t know if this is just a printer profile problem. When I am done with the current project I’ll switch it to another “supported” printer. I’m just curious why it’s substituting these layers of different heights in the code and if that’s the cause of my issues. I’ll do some more digging on this side and post back.