Ultimaker/Cura #18880: quest_2_to_quest_3_strap_adapter.stl

frozentoasty, 2024-04-11:
### Cura Version

5.6

### Operating System

Windows 11

### Printer

Anycubic Vyper

### Name abnormal settings

normal

### Describe model location

its in the center

### Describe your model

its a meta quest 2 to quest 3 headstrap adapter 

### Add your .zip here ⬇️

[image]
[AV_quest_2_to_quest_3_strap_adapter.zip](https://github.com/Ultimaker/Cura/files/14950711/AV_quest_2_to_quest_3_strap_adapter.zip)

HellAholic, 2024-04-11:
@frozentoasty give the 5.7 version a try, there were a few improvements to tree support and slicing performance. I can slice your model in 5.7 without issue.
[image]

frozentoasty, 2024-04-14:
Yes i did the same and it worked. Thank you very much for your answer!

closed 2024-04-26 by HellAholic