Ultimaker/Cura #18882: importing files takes really long without indication of it doing anything

DowneyX, 2024-04-12:
### Cura Version

5.7

### Operating System

arch linux

### Printer

ender 3 v2 neo

### Reproduction steps

1: open cura.
2: click on folder icon
3: select stl

### Actual results

there is no prompt telling me that it is working on importing my model. it takes a full 3 minutes to load a simple STL file.

### Expected results

it should prompt me that it is importing my file. 
it should also take less than 5 seconds to import a simple STL file

### Add your .zip and screenshots here ⬇️

looking at the logs it seems that the issue stems from the USBPrinting.AutoDetectBaudJob job. it is trying to establish serial connections with all my tty ports (and failing). after this process is done it imports my STL file.

[stderr.log](https://github.com/Ultimaker/Cura/files/14956345/stderr.log)

DowneyX, 2024-04-12:
a work around for now is to disable the USB printing plugin

nallath, 2024-04-15:
Oh wow, that is a _lot_ of usb ports that it's trying to poll. 

If you don't want usb printing, the workaround is indeed to disable the usbprinting plugin

olekolek1000, 2024-05-24:
Can confirm.

Din0v, 2024-08-30:
wow, thanks that worked i am on version 5.8.1

Jooosl, 2024-09-01:
hey i don't have any usb printing plugin installed in the marketplace, is it installed by default?
still having the issue

TDB-ev, 2024-09-28:
> > hey i don't have any usb printing plugin installed in the marketplace, is it installed by default? still having the issue
> 
> I was also confused, but here's how to disable: Open Cura, click "Marketplace" top right; in new window click gear icon top right; scroll down a lot (about 1/3 of the scroll bar) and disable "USB Printing". It is (I think) a default extension.

This works for me, thanks!

Gh0stX4, 2025-09-13:
> a work around for now is to disable the USB printing plugin

This also fixes my problem but for what ever reason Cura was working fine until one day for some reason but what ever is going on should probably be fix as soon as possible or some confused person might think their models aren't loading and give up (because I was that person at one point) or maybe even give false bug reports (that there models aren't loading)

Cura version: 5.10.2
Operating System: Arch Linux