Ultimaker/Cura #18887: Cura crashes a few seconds after opening on MacOS Sonoma 14.2.1

clippycoder, 2024-04-12:
### Cura Version

5.7

### Operating System

MacOS Sonoma 14.2.1

### Printer

Creality Ender 3 S1

### Reproduction steps

I tried to open Cura. It would load as normal, but a few seconds after opening it would just quit, without any "Cura has crashed unexpectedly" popup from the OS, which is unusual. The issue was persistent through multiple attempts as well as restarting.

### Actual results

Cura opens, and the closes.

### Expected results

Cura should have opened

### Add your .zip and screenshots here ⬇️

[cura.log](https://github.com/Ultimaker/Cura/files/14963802/cura.log)

HellAholic, 2024-04-12:
Hey, thanks for the report.
Do you by any chance have the printer connected via USB to your Mac device?
If not, try disabling the USB printing plugin. You can do this via configuration folder where you got the cura.log file.
There is a plugins.json file there that you can edit to `{"disabled": ["USBPrinting"], "to_install": {}, "to_remove": []}`

clippycoder, 2024-04-14:
I don't use Cura with the printer connected via USB, I always save the gcode to an SD card (which wasn't attached, either).