Ultimaker/Cura #18890: undo for changing printing parameters

illusionmanager, 2024-04-13:
### Is your feature request related to a problem?

I'm playing with printer parameters and watch how it slices, but sometimes I want do undo several changes (which I didn't all remember). Hitting the undo button, only undoes the editing of the model.

### Describe the solution you'd like

let the undo button work also work with parameter changes.

### Describe alternatives you've considered

there is no alternative besides creating a profile for every step which would be ridiculous.

### Affected users and/or printers

everyone will benefit.

GregValiant, 2024-04-14:
I have found myself in that position as well.  I have four base "custom" profiles and I usually end up resetting whichever one I started with back to it's saved defaults and then I start over.  Since I made a hash of the first attempt, I have some idea of where I want to go on the next attempt.  It's usually relatively painless.

It would seem that on every setting change an undo file would need to be created.  I suppose it could just contain the last setting that was changed, but it might have to keep track of all 600 settings.  Using a different "undo" key combination might be necessary as Ctrl-Z is dedicated to the workspace.  All the Undo files would need to be deleted at some point (I would guess when Cura is closed).

@fieldOfView is this something you have thought about at all?  Is it something a plugin could do?

illusionmanager, 2024-04-14:
you don't need many undo files. you just keep the changes in memory, and write whatever you want to store in case of a crash (1 status file). A list of changes takes very little memory, so that's no problem. Nobody expects undo's to be available when you restart cura.