Ultimaker/Cura #18891: View port doesn't render on wayland

uwumdoescomputershit, 2024-04-14:
### Cura Version

5.7.0

### Operating System

Arch Linux, Hyprland (wayland), nvidia gtx 1060

### Printer

Creality Ender 5

### Reproduction steps

1. Launch the cura flatpak in a wayland compositor (in my case hyprland)

### Actual results

All UI elements render fine apart from the view port in the center. When running the application under xwayland it works fine.

### Expected results

All UI items render fine (like they do on xorg).

### Add your .zip and screenshots here ⬇️

[image]

nallath, 2024-04-15:
Does this also happen for the appimage? We don't officially support the flatpack, so there might be subtle differences between the two that cause this issue.

FlorianFlatscher, 2024-04-18:
Same here. With official AppImage on fedora 39

robertaramar, 2024-04-25:
Same here. With official AppImage 5.7.1, with rpm version 5.5.0 and flatpak 5.7.1, both on Fedora 39 and Fedora 40

xubz, 2024-04-28:
Arch Linux with Nvidia/KDE Plasma 6/Wayland

Same issue. Works with these arguments **-platform xcb -platformtheme gtk3** but it's a flickery mess. I guess, I have to wait for the much hyped "Explicit Sync" patch. Atleast can slice and print!

Tabisch, 2024-04-28:
Had the same issue on Fedora 39 and 40 with Flatpak.

Setting QT_QPA_PLATFORM=xcb in Environment in FlatSeal fixed it for me.
For me it runs smooth on Fedora 40 with Gnome 46.1.

robertaramar, 2024-04-28:
It's nowhere near to "smooth". It does run, however, it flickers a lot, and sometimes I have a white window that only refreshes when I rotate or zoom.
In addition, it takes ages until the model gets a first rendering.
I did not experience this when I was running my desktop in an X session. Only after I moved to Wayland.

sausix, 2024-04-29:
Good news everyone! Not a fix but a non flickery workaround.

According to this hint I tried to tweak EGL which raised an error before:
https://wiki.archlinux.org/title/OpenGL#OpenGL_over_Vulkan_(Zink)

At least 5.7.1 seem to run fine in pure Wayland on my Nvidia RTX 4060 and `nvidia-dkms`.

Running cura with these two environment variables seem to be enough for me:
`__EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json MESA_LOADER_DRIVER_OVERRIDE=zink cura`

You'll probably also need `QT_QPA_PLATFORM="wayland;xcb"` which is set by my global env file which I share for free too:

```bash
# /etc/profile.d/wayland.sh 
# Tweaks mostly from: https://wiki.archlinux.org/title/Wayland
export QT_QPA_PLATFORM="wayland;xcb"
export SDL_VIDEODRIVER="wayland,x11"
export CLUTTER_BACKEND=wayland

export GBM_BACKEND=nvidia-drm
export __GLX_VENDOR_LIBRARY_NAME=nvidia

export ELECTRON_OZONE_PLATFORM_HINT=wayland

# GTK to X11
#GDK_BACKEND=x11

# Thunderbird and Firefox
export MOZ_ENABLE_WAYLAND=1

# LibreOffice
# /etc/profile.d/libreoffice-still.sh
export SAL_USE_VCLPLUGIN=qt6

# Java blank screen?
#export _JAVA_AWT_WM_NONREPARENTING=1
```
Rule of thumb for NVidia and Wayland: No xwayland.

basilverthorn, 2024-04-30:
The above seem to fix it temp at least thank you. I can not use it per say. What ever build Qidiprint is on it just works in Wayland, Hperland etc.. Its an older version of Cura though thats been reskined.

sausix, 2024-04-30:
@basilverthorn
I never had flickering in pure Wayland. The print bed and some UI was missing. Old workaround was using xwayland at the cost of flickering within Cura.

Do you run Cura in a pure Wayland session?
`QT_QPA_PLATFORM=wayland cura` and add the Vulkan fix variables to get the bed and UI visualized.
Then cura or the Qt backend will truely use Wayland.

basilverthorn, 2024-04-30:
Yes true Wayland or I run Hyprland. Migrating to Hyprland, but see the same thing in both. I will try this tomorrow if I can, working late today unfortnately.

basilverthorn, 2024-05-02:
So I tried the vulkan fix and QT_QPA and I know recieve: "Could not probe OpenGL. This program requires OpenGL 2.0 or higher. Please check your video card drivers." Trying to start Cura, the appimage runs, but does not have a build plate.

basilverthorn, 2024-06-19:
Still an issue anyone come up with solution? No build plate still.

sausix, 2024-06-19:
Mesa seems to be broken on EGL/OpenGL at Qt level already:
https://gitlab.freedesktop.org/mesa/mesa/-/issues/11316

How did you get mesa 24.1.1 early on May 2th? Even Arch users got that version on June 6th.
Can you downgrade mesa on your distribution to 24.1.0? We didn't find a better fix or workaround yet.

42Willow, 2024-07-27:
I am also on Hyprland but installed cura via the aur `cura-bin` package. I am also using intel integrated graphics. I am not experiencing this issue but I get a segfault when trying to open a file.

DeBeerBuilds, 2024-08-25:
> Same here. With official AppImage 5.7.1, with rpm version 5.5.0 and flatpak 5.7.1, both on Fedora 39 and Fedora 40 Only chance to "somewhat" run it is to set QT_QPA_PLATFORM=xcb before running it.

How do you check that?

HenrikElofsson, 2024-08-30:
I have the same issue with AppImage 5.8.1. It's a Gentoo system with Nvidia graphics.

mentasuave01, 2024-10-12:
same issue here

KyleStilkey, 2024-10-12:
Testing @wawanbreton Wayland fix on Bazzite and it's so far working fine here. I can now see it rendering the main image again.