Ultimaker/Cura #18892: Fail to import STL files

pogechi, 2024-04-14:
### Cura Version

5.7.0

### Operating System

Linux

### Printer

Anycubic Kobra NEO

### Reproduction steps

Open File
From Disk
Select STL file

### Actual results

Nothing gets imported

### Expected results

Import STL file

### Add your .zip and screenshots here ⬇️

Nothing to show

nallath, 2024-04-15:
Can you please provide the log files? You can find information where to find them here: https://github.com/Ultimaker/Cura/wiki/Reporting

nilesjohnson, 2024-04-17:
I'm having a similar problem; also linux, but with a cura printer.  The issue is new with some update in the past few months.  Actually, after waiting ~5 minutes, the stl does finally load, and then I can load others instantly.

Searching around, I found [this rambling thread](https://community.ultimaker.com/topic/44214-cura-doesnt-show-stl-models/) that describes similar behavior in September 2023.  It concludes with a cryptic comment to disable usb printing.  I tried that and, it works!  I don't have a printer connected by usb to the computer I use for slicing.

When I check the log files from before I had the plugin disabled, I see thousands of debug messages like the following.  Note the timestamps are separated by `,001` seconds or less!?

```
2024-04-17 05:27:20,090 - DEBUG - [MainThread] cura.TaskManagement.OnExitCallbackManager.triggerNextCallback [53]: Scheduled the next on-app-exit callback [<bound method USBPrinterOutputDevice._checkActivePrintingUponAppExit of <USBPrinting.USBPrinterOutputDevice.USBPrinterOutputDevice object at 0x71eb78289510>>]
2024-04-17 05:27:20,090 - DEBUG - [MainThread] cura.TaskManagement.OnExitCallbackManager.triggerNextCallback [53]: Scheduled the next on-app-exit callback [<bound method USBPrinterOutputDevice._checkActivePrintingUponAppExit of <USBPrinting.USBPrinterOutputDevice.USBPrinterOutputDevice object at 0x71eb63f291b0>>]
2024-04-17 05:27:20,091 - DEBUG - [MainThread] cura.TaskManagement.OnExitCallbackManager.triggerNextCallback [53]: Scheduled the next on-app-exit callback [<bound method USBPrinterOutputDevice._checkActivePrintingUponAppExit of <USBPrinting.USBPrinterOutputDevice.USBPrinterOutputDevice object at 0x71eb78289b40>>]
2024-04-17 05:27:20,091 - DEBUG - [MainThread] cura.TaskManagement.OnExitCallbackManager.triggerNextCallback [53]: Scheduled the next on-app-exit callback [<bound method USBPrinterOutputDevice._checkActivePrintingUponAppExit of <USBPrinting.USBPrinterOutputDevice.USBPrinterOutputDevice object at 0x71eb78289480>>]
2024-04-17 05:27:20,091 - DEBUG - [MainThread] cura.TaskManagement.OnExitCallbackManager.triggerNextCallback [53]: Scheduled the next on-app-exit callback [<bound method USBPrinterOutputDevice._checkActivePrintingUponAppExit of <USBPrinting.USBPrinterOutputDevice.USBPrinterOutputDevice object at 0x71eb63f29bd0>>]
2024-04-17 05:27:20,091 - DEBUG - [MainThread] cura.TaskManagement.OnExitCallbackManager.triggerNextCallback [53]: Scheduled the next on-app-exit callback [<bound method USBPrinterOutputDevice._checkActivePrintingUponAppExit of <USBPrinting.USBPrinterOutputDevice.USBPrinterOutputDevice object at 0x71eb78289000>>]
2024-04-17 05:27:20,092 - DEBUG - [MainThread] cura.TaskManagement.OnExitCallbackManager.triggerNextCallback [53]: Scheduled the next on-app-exit callback [<bound method USBPrinterOutputDevice._checkActivePrintingUponAppExit of <USBPrinting.USBPrinterOutputDevice.USBPrinterOutputDevice object at 0x71eb782896c0>>]
2024-04-17 05:27:20,092 - DEBUG - [MainThread] cura.TaskManagement.OnExitCallbackManager.triggerNextCallback [53]: Scheduled the next on-app-exit callback [<bound method USBPrinterOutputDevice._checkActivePrintingUponAppExit of <USBPrinting.USBPrinterOutputDevice.USBPrinterOutputDevice object at 0x71eb78289630>>]
2024-04-17 05:27:20,092 - DEBUG - [MainThread] cura.TaskManagement.OnExitCallbackManager.triggerNextCallback [53]: Scheduled the next on-app-exit callback [<bound method USBPrinterOutputDevice._checkActivePrintingUponAppExit of <USBPrinting.USBPrinterOutputDevice.USBPrinterOutputDevice object at 0x71eb78289750>>]
2024-04-17 05:27:20,092 - DEBUG - [MainThread] cura.TaskManagement.OnExitCallbackManager.triggerNextCallback [53]: Scheduled the next on-app-exit callback [<bound method USBPrinterOutputDevice._checkActivePrintingUponAppExit of <USBPrinting.USBPrinterOutputDevice.USBPrinterOutputDevice object at 0x71eb782897e0>>]
2024-04-17 05:27:20,093 - DEBUG - [MainThread] cura.TaskManagement.OnExitCallbackManager.triggerNextCallback [53]: Scheduled the next on-app-exit callback [<bound method USBPrinterOutputDevice._checkActivePrintingUponAppExit of <USBPrinting.USBPrinterOutputDevice.USBPrinterOutputDevice object at 0x71eb78289870>>]
2024-04-17 05:27:20,093 - DEBUG - [MainThread] cura.TaskManagement.OnExitCallbackManager.triggerNextCallback [53]: Scheduled the next on-app-exit callback [<bound method USBPrinterOutputDevice._checkActivePrintingUponAppExit of <USBPrinting.USBPrinterOutputDevice.USBPrinterOutputDevice object at 0x71eb78289990>>]
2024-04-17 05:27:20,093 - DEBUG - [MainThread] cura.TaskManagement.OnExitCallbackManager.triggerNextCallback [53]: Scheduled the next on-app-exit callback [<bound method USBPrinterOutputDevice._checkActivePrintingUponAppExit of <USBPrinting.USBPrinterOutputDevice.USBPrinterOutputDevice object at 0x71eb78289900>>]
2024-04-17 05:27:20,093 - DEBUG - [MainThread] cura.TaskManagement.OnExitCallbackManager.triggerNextCallback [53]: Scheduled the next on-app-exit callback [<bound method USBPrinterOutputDevice._checkActivePrintingUponAppExit of <USBPrinting.USBPrinterOutputDevice.USBPrinterOutputDevice object at 0x71eb78289a20>>]

```

pogechi, 2024-04-17:
Thank you Niles, your suggestion seems to have worked for me as well.

taleroangel, 2024-04-18:
I am experiencing the same issue with Cura 5.7.0 (Flatpak) with Linux 6.8.6, booting from an old kernel version (6.7.9) fixes the issue. Might be a kernel compatibility issue