Ultimaker/Cura #18893: anti aliassing option

illusionmanager, 2024-04-14:
### Is your feature request related to a problem?

when printing slanted surfaces the infamous staircase effect is almost always visible.
Generating in between lines at half the height or even quarter height to make the step less pronounced seems like an option worth investigating.

### Describe the solution you'd like

This will generate the desired anti aliassing effect.

### Describe alternatives you've considered

The alternative is slicing the whole object at tiny layer heights (currently doing 0.04mm layers. But that creates a lot of pressure in the hot end and increases the print time by a factor 4 (comparing 0.04mm vs 0.12 mm for the normal height of a detailed object)

### Affected users and/or printers

this will benefit all users.

HellAholic, 2024-04-15:
Hey @illusionmanager ,
Thanks for the suggestion but unfortunately this is not feasible. Any attempt to add middle steps as you suggest will result in lowering the layer height.
You can alter the slicing tolerance to fit your needs, but adding in between lines means you need to add a layer, which will result in lower layer height.
[image]

The other option you might be referring to is "use adaptive layers" which is under experimental as a setting in current Cura.

GregValiant, 2024-04-15:
@illusionmanager No fair using a 2d image for a 3d print.
What does the rest of the model look like?  That layer "step" needs to go all around and all the way through.  It can't just be for a specific part of a particular outer wall extrusion.  There isn't a way to simply pick out that one particular piece of one particular feature.
Cura works by layers.  Any individual slice doesn't know what happened on the previous slice and can't know what will happen on the next slice (because it hasn't occurred yet).

It's the nature of FDM that there will be steps.  This is at a layer height of 0.05.
[image]

GregValiant, 2024-04-15:
This is with Adaptive Layers turned on.  Layer height is 0.2 and Max Variation for Adaptive Layers is 0.18 so the curve is at 0.02 layer height.  The print time did almost double.
[image]

illusionmanager, 2024-04-15:
well, look how smooth that is. Wouldn't you like that almost for free, instead of doubling the time? The strong disadvantage of adaptive layers is that it print the full layer at such tiny heights. The the suggested anti aliasing solution would only fill in the steps between the normal layers, which is just a tiny amount. Also many objects (including those that people deliberately print at an angle for strength reasons), have slanted surfaces from top to bottom, making the adaptive layer solution super slow.

I'm almost temped to manually add some gcode, just to show that it would work great, but that would be a lot of work.

GregValiant, 2024-04-15:
I just don't know how to look at this layer and decide where a step should be made.  I can see where line width would need to be adjusted as well if the width of the step was some in-between number.
[image]

If it's even possible, it looks like a lot of work.

illusionmanager, 2024-04-15:
send me the stl of your object and I will indicate it for you.

Actually I don't think it is too hard to code at all. I used to code a long time ago, but ended up telling other people what to code :-) I have a reasonably good feeling for this. One thing is for sure developers love working on new stuff instead of fixing existing problems, so they might really like it.

Part of the code is already there in the experimental slicing tolerance option. It already detects which layer lines are fully inside the model, fully outside, or partially inside. That amount of "partiality" determines the amount you can fill using anti aliasing. For 2x anti aliasing, you'll look for those line parts that are 50% or more filled, but not 100% and you print those with 50% layer height and the rest at 100%. That's ok for layers on top, for overhangs you'll first print the 100% part, followed by the 50% part.

 I could say "that's it" and you could  just implement it like that, but it could be more precise, because the 50% mark might not be at the exact multiple of the line width. Assume a step that would normally be 3 line wide, the 50% mark would be at 1.5 line, so that's where you print the layer at 50% height, you would need 2 lines of 50% height. The 100% fill boundary would however partially overlap this second line, so when you print that outer wall at full height, it is already for 25% filled by the half height line. Therefore you print that specific line with 75% extruded material. 
 That might be a bit hard to understand without a good example. So if you need that, I could spend time making that.

GregValiant, 2024-04-15:
Here's the model.
[junk.zip](https://github.com/Ultimaker/Cura/files/14984326/junk.zip)
The code will be in Cura Engine.  Most of it is C++ I think.  There may be some Python mixed in.

GregValiant, 2024-04-16:
The coding is way beyond me.  I would have no idea how to pick that line to be at half height rather than one of the other lines.

illusionmanager, 2024-04-16:
luckily that part is already inside the code :-). I didn't draw those lines myself, I let cura do it for me. An alternative way to generate those lines is to look at the difference between "middle" and "inclusive". That difference is composed of those lines. 

Basically lines that are more than 50% filled but not fully 100%