Ultimaker/Cura #18895: Slicing Failed on Print-in-Place Bearing

okokel, 2024-04-15:
### Cura Version

5.7.0

### Operating System

Windows 10

### Printer

Ender-3 V2

### Name abnormal settings

Using default Standard Quality settings for 0.4mm nozzle, 0.2mm layer height

### Describe model location

Center of bed

### Describe your model

3d-printable bearing, multiple parts that are very close but do not overlap.  Slices fine in Cura 5.6

### Add your .zip here ⬇️

[Slicing Failure Cura 5.7.zip](https://github.com/Ultimaker/Cura/files/14978956/Slicing.Failure.Cura.5.7.zip)

GregValiant, 2024-04-15:
Thanks for the report.
I didn't have any trouble on my Win 10 system.
Try right-clicking on the Cura icon on your desktop and select "Run as Administrator".  Let us know if that makes a difference.  

In the meantime I'll mark this as a duplicate.

jellespijker, 2024-04-23:
Thanks for reporting this, when I try slicing this in our upcoming 5.7.1 release, it slices with out problem. Closing this issue as fixed
[image]

closed 2024-04-23 by jellespijker