Ultimaker/Cura #18903: Camera mounts

J0hnEastwooD, 2024-04-16:
### Cura Version

5,7,0

### Operating System

Windows 10

### Printer

Flying bear ghost 6

### Name abnormal settings

Since the 5.6 update, there have been such failures with most stl files

### Describe your model

Camera mounts to look under the ice

### Add your .zip here ⬇️

[Kreplenie_kameri v1.zip](https://github.com/Ultimaker/Cura/files/15000308/Kreplenie_kameri.v1.zip)

GregValiant, 2024-04-16:
Thanks for the report.
I didn't have any trouble slicing this and I haven't been experiencing any problems using 5.7.0.
One thing has come up and it might affect this, right-click on the Cura icon on your desktop and select "Run as Administrator" and see if it makes a difference slicing.  Let us know whether it helps or not.

J0hnEastwooD, 2024-04-18:
The problem has not gone away

J0hnEastwooD, 2024-04-18:
Another problem, slicing happens when you change any parameter in the print settings, also appeared after 5.6

J0hnEastwooD, 2024-04-18:
Small files are sliced, but there are no more, under 1 condition if you change any parameter

jellespijker, 2024-04-23:
can you share the 3mf project file instead of the stl, we need your used printers and settings to reproduce the issue

J0hnEastwooD, 2024-04-27:
https://drive.google.com/file/d/14-JmiapqdeUiNFBvzD063T28mtL26hv0/view?usp=drive_link
this model is not cut, under any settings, if I understood correctly what kind of file you need, then it is on the link, I cannot cut it at all, and I managed to print the file on the topic

GregValiant, 2024-04-28:
@J0hnEastwooD 

- Load a "problem" model into Cura.  Set Cura up to slice.  (Jellespijker will need the error reproduced.)
- Select the command "File | Save Project".  That will create a special 3mf file that contains the model, your printer, and your settings.
- In Windows "File Explorer" right-click on the file and select "Send To" and then "Compressed (zipped) folder".
- Back in Cura, slice the project (to create the error).  Use the "Help | Show Configuration Folder" command.  In your configuration folder will be a file "Cura.log".  Please copy and paste that file into the zip folder.

Github places a limit on the size of the files that may be posted here.  If the size of that zip folder is less than 25mb you can post it here.  If it is larger than 25mb then you can use Google Drive but you must "Share" the file or we don't have permission to download it.

J0hnEastwooD, 2024-04-28:
[cura.log.zip](https://github.com/Ultimaker/Cura/files/15142214/cura.log.zip)

https://drive.google.com/file/d/103avXI23MlAW6I2prC0WUxp4YdKdGErD/view?usp=drive_link

GregValiant, 2024-04-28:
Thanks for that.

This took a long time to slice, but it did finish.  This is in 5.7.1 on a Win10 system.
[image]

J0hnEastwooD, 2024-04-28:
Well, they've been sliced for a long time before. As I understand it, I have no choice how to reinstall Windows completely?

J0hnEastwooD, 2024-04-28:
Well, he cut the model for a long time before that. As I understand it, I have no choice how to reinstall Windows completely?

GregValiant, 2024-04-28:
Wait for Jellespijker to get back.  He is on the Cura team.

J0hnEastwooD, 2024-04-28:
It turned out awkwardly, I have already demolished 5.4.0 on which I was typing, installed 5.7.1.

J0hnEastwooD, 2024-04-28:
5.7.1 does not cut either

HellAholic, 2024-04-29:
Hey @J0hnEastwooD 
There is a workaround to remove all the settings/profiles/printers/plugins/etc and seeing if that fixes the issue.
This would mean that you would loose all your custom profiles and settings, so make sure you have a backup before attempting this.

What you can do:

1. Go to the %appdata%/Roaming/cura
2. Back up the contents (any version you see a folder in there 4.x or 5.x) -> make a zip and store somewhere as backup
3. Remove all the contents of the %appdata%/roaming/cura/ (any version 4.x or 5.x)
4. Start Cura again
5. Add your printer
6. Attempt to slice
7. If the process didn't succeed, you can restore from the backup.

J0hnEastwooD, 2024-04-29:
[cura.log](https://github.com/Ultimaker/Cura/files/15148213/cura.log)
It didn't help

HellAholic, 2024-04-29:
I can see 4 sentry issues reported in line with your crashes in the logs (rough guesstimation based on printer, OS, cura version):
- CURAENGINE-3Y
- CURAENGINE-3C
- CURAENGINE-6F
- CURAENGINE-5X

J0hnEastwooD, 2024-04-29:
I didn't understand anything, but it's very interesting

jellespijker, 2024-04-29:
That basically means that we captured detailed information regarding your crash, which is the first step into determining how often it occurs, how many other user experience a similar crash and it can help us fix it.

J0hnEastwooD, 2024-05-09:
I am haunted by this problem, I have just noticed that if you move the model (now this problem is with almost all high models) in any direction by a certain number of mm, then the slicer cuts, what can this be related to?

closed 2024-05-31 by J0hnEastwooD