Ultimaker/Cura #18906: Slicing failed

oliof, 2024-04-17:
### Cura Version

5.7.0

### Operating System

POP!_OS Linux 20.04

### Printer

Betaminion (custom cartesian printer)

### Name abnormal settings

Tree Support

### Describe model location

center of bed, moving the model does not work

### Describe your model

Custom model created in FreeCAD. Model is watertight according to Meshtools Plugin analysis

### Add your .zip here ⬇️

canetop.zip is the project file
canetopzipfcstd.zip is the FreeCAD source model
canetopstl.zip is the STL as exported by FreeCADs export function

[canetop.zip](https://github.com/Ultimaker/Cura/files/15009173/canetop.zip)
[canetop-fcstd.zip](https://github.com/Ultimaker/Cura/files/15009181/canetop-fcstd.zip)
[canetopstl.zip](https://github.com/Ultimaker/Cura/files/15009194/canetopstl.zip)

oliof, 2024-04-17:
Curiously I got a model error pop up just now, while zooming into and out of the model. 

[image]

Looking at the model within FreeCAD Mesh workbench, it looks like the thread normals are kinda weird, but FreeCAD itself seems to fail fixing it. I am trying alternate meshing methods now to see if I can get better results.

oliof, 2024-04-17:
Tried with a higher resolution mesh (440MB, so not attached here) and the same error occurs.

oliof, 2024-04-17:
With some help I improved the FreeCAD model, but the STL still does not slice in Cura (but slices in other slicers).
[canetop.zip](https://github.com/Ultimaker/Cura/files/15014638/canetop.zip)

oliof, 2024-04-18:
It looks like I have found a way to confuse both FreeCAD and Cura with the same model. I am addressing the issue on the FreeCAD side as well. Here are some images showing how the model appears, and how a high resolution mesh interprets the available model data:

[image]

[image]

GregValiant, 2024-04-18:
Thanks for the report.
Good job confusing both apps.  I can usually only manage one at a time.
It sliced fine on my Win10 system and Cura 5.7.0.  I know that's no help but there it is.
I checked the model with MS 3D Builder and it doesn't show any errors.  I also checked it with an online repair surface and it didn't find anything wrong either.  I'm not sure what Mesh Tools found that it didn't like.
In regards to the resolution, the model in the project file you posted has ~135,000 faces.  That is reasonable.
This is not the first report regarding problems on a Linux system.  The Cura team will take a look.
Have you tried moving the model on the build plate?

Here it is sliced.  I actually have one similar to this on a "pool cue walking stick" that dates to the early 1950's.
I think this would turn out a lot better at .14 layer height and maybe with Adaptive Layers enabled.
[image]

oliof, 2024-04-18:
I have redone this model multiple times in FreeCAD now because it just wasn't right. I also re-exported it with reasonable and unreasonable amounts of mesh resolution (354MB binary STL being the largest).
 
The model passes FreeCAD's geometry check now, and still FreeCAD's own Mesh Analyzer finds "folds on surface" close to the taper of the thread:
 
[image]

but no other mistakes. These folds happen irrespective of the meshing method used (FreeCAD native, netgen, gmsh, mefisto), and simply increase in number with higher resolution meshes. Unfortunately I don't know what constitutes a 'fold' here. 

Other methods of checking the model for errors (Blender, Grid:Space's mesh:tool) also found no issue with the mesh. 

I did try moving the model on the build plate, both manually, and with MeshTool's random placement method, to no avail. I even tried rotating the part and cutting it with the banana slice tool, and that also didn't yield good results.

I've added my most recent version of this piece below, now with proper threads and a little offset to the lip, but still with those fold errors.

[canetop-latest.zip](https://github.com/Ultimaker/Cura/files/15031218/canetop-latest.zip)

GregValiant, 2024-04-19:
I know you have uploaded a bunch of files but one more is in order.
Slice the model, get the error, and then use the "Help | Show Configuration Folder" command.  Zip the "Cura.log" file and post the zip folder here.  Maybe it will give a clue as to what is causing the crash/non-slice.

oliof, 2024-04-19:
D'oh, how could I have forgotten the log file ...

Just a note that with the AppImage and Linux conventions, the config folder lives in `$HOME/.config/cura/$VERSION`, but the cura log lives in `$HOME/.local/share/cura/$VERSION`. But here goes.

I did the following:

1. Quit Cura
2. Removed the old cura.log file
3. Started Cura
4. Loaded the canetop-hole-Body.stl
5. Selecting the Built-In 'Draft' print profile, hit 'Slice'
6. Almost _immediately_ the error message pops up
7. Quit Cura
8. Zipped up the log file and STL to attach here.

[cura-log-and-stl.zip](https://github.com/Ultimaker/Cura/files/15033512/cura-log-and-stl.zip)

oliof, 2024-04-19:
... and looking into the logs myself, it seems to be an issue with the TiledInfill plugin. I cross-checked by trying to slice another file that sliced before, which immediately failed, then uninstalled the TiledInfill plugin, and was able to slice the canetop model without issue.

I think it's still worth investigating why the TiledInfill plugin fails. Glad to assist with testing on Linux for that!

GregValiant, 2024-04-21:
It does look like TiledInfill is an issue on your system.

I'm not sure, but the "folds" at the top of the thread might be because it is coming to a knife edge there.  The rest of the thread has a flat on top but that clearance between the thread and the shoulder gradually reduced the flat to 0 width and the knife edge developed.

@HellAholic here's a different kind of failure.

HellAholic, 2024-04-21:
Hey @oliof,
the engine plugins have their own separate logs so unfortunately the important bits are hiding in there. However, Linux has been a bit unfair towards the tiled infill plugin in particular, see discussion in: https://github.com/Ultimaker/CuraEngine_plugin_infill_generate/issues/7

I've had the similar immediate fail on the linux and the proposed solution fixed the issue for me.
Give that one a try, if it doesn't work out please include the log from the plugin as well.

Cheers!

oliof, 2024-04-21:
Hi,
I don't need the Tiled Infill plugin for now, so for me it's fine to disable it for the moment. I will keep the proposed workaround in mind.

Feel free to close this issue.

closed 2024-04-24 by GregValiant