Ultimaker/Cura #18908: slicing failed

Govardhan3d, 2024-04-17:
### Cura Version

5.7.0

### Operating System

Windows 11

### Printer

creality ender-3 ender-3 v2

### Name abnormal settings

yes its in default profiles

### Describe model location

center of buildplate

### Describe your model

After slicing and it gets failed and i tried to mesh fix too

### Add your .zip here ⬇️

[image]
[CE3E3V2_Body1.zip](https://github.com/Ultimaker/Cura/files/15013189/CE3E3V2_Body1.zip)

chross, 2024-04-19:
Same problem here.
Using the Linux x64 AppImage on Ubuntu 22.04 in a VM on Windows 11 host.

jellespijker, 2024-04-23:
The zip file contains GCode which seems to indicate that slicing succeeded.
Can you please share the 3mf project file from the actual failed slice

closed 2024-05-07 by github-actions[bot]