Ultimaker/Cura #18914: Bug while slicing one of many files in a multi part print

MidnightSkeptik9, 2024-04-18:
### Cura Version

5.7.0

### Operating System

windows 11

### Printer

Ender 3 v3 KE, (modified ender 3 pro profile)

### Describe model location

It slices if i rotate it so its in its original orientation, vertical, i rotated it so that the flat face is on the build plate and now it wont slice.

### Describe your model

I have never sliced it before

### Add your .zip here ⬇️

[CE3PRO_Hull_Top.zip](https://github.com/Ultimaker/Cura/files/15031103/CE3PRO_Hull_Top.zip)

GregValiant, 2024-04-19:
Thanks for the report.
I didn't have any trouble slicing the project file.
Use the "Help | Show Configuration Folder" command, zip the Cura.log file and post the sip here.  Maybe it will show what is happening on your system.

jellespijker, 2024-04-23:
Thank you for reporting this.

When I try slicing it in our upcoming 5.7.1 release, it slices as expected. closing this issue as fixed.

[image]

closed 2024-04-23 by jellespijker