Ultimaker/Cura #18915: slicer won't print thin walls

delta1981, 2024-04-19:
### Cura Version

5.7.0

### Operating System

windows 11

### Printer

creality ender 2 pro

### Reproduction steps

I just sliced the file as normal with the settings I used in the last version of cura, 3 walls, cubic infill, 50mms print speed, skirt print bed adhesion

### Actual results

when printed there are gaps where there should be solid walls
[image]

### Expected results

this is what it should look like when sliced,
[image]
 this one was done on creality print and matches my previous print results with cura 5.6.0

### Add your .zip and screenshots here ⬇️

here is the file as requested above along with a screen shot 
[image]
[CE2_Flame light back fence.zip](https://github.com/Ultimaker/Cura/files/15039490/CE2_Flame.light.back.fence.zip)

GregValiant, 2024-04-19:
Thanks for the report.
In the Wall Settings is "Print Thin Walls".  It is not enabled in your project file and I get the same results that you show.

When I enable "Print Thin Walls" this is the result in 5.7.0
[image]

This looks like a settings issue.  I'll remove the bug label for now and leave this open so you can respond.

delta1981, 2024-04-19:
this has fixed the issue don't know why it became unchecked when I upgraded from 5.6.0 to 5.7.0 hanks for all the help and the quick response

closed 2024-04-19 by delta1981

GregValiant, 2024-04-19:
In the printer definition files for both 5.6 and 5.7 it is set to "false".  Maybe you had an override in a custom profile.