Ultimaker/Cura #18917: Add + - buttons in the Preview mode to change the layer height one by one

PeterVH72, 2024-04-20:
### Is your feature request related to a problem?

When, in Preview, when i want to see if a certain part of the model needs support or not, i want to change the layer height one by one. You can use the vertical scrol bar to change the layer but when there are a lot of layers, changing the layer height one by one with the scroll bar is to difficult. I always have to type in the number of the layer when i want to change it one by one. 
Adding plus minus buttons under the scroll bar would be a lot easier.

### Describe the solution you'd like

Adding plus minus buttons under the scroll bar would be a lot easier to change the layer. 
Clicking the + or - button changes the layer view by 1 layer height.

### Describe alternatives you've considered

No alternatives

### Affected users and/or printers

I think that everyone who prints detailed models with supports would love this!

### Additional information & file uploads

[image]

gh13, 2024-04-20:
Once you've selected the layer sllider you can also use the up & down arrow keys to change layer one by one

Asterchades, 2024-04-21:
At least in the Windows client you can also hold down Control (Ctrl) and use Up and Down at any time. It's probably the same in other OSs but I have no facilities for testing that.

Though I can still see the value in the suggestion for tablets, which lack a physical keyboard.

gh13, 2024-04-21:
I didn't know about the Ctrl + Up and Down shortcut. You learn something everyday. 😊
Do you know if there's a list of Cura keyboard shortcuts anywhere

Asterchades, 2024-04-21:
>  Do you know if there's a list of Cura keyboard shortcuts anywhere

Not that I'm aware of, though that of course doesn't mean it doesn't exist. It really does strike me as the kind of thing that should be collated somewhere easily accessible as there are a number of shortcuts available, such as T to enable the Move tool (M is used for Mirror) - though at least you can "discover" those by holding the mouse over their respective icons and waiting for the tooltip to show.

gh13, 2024-04-21:
Thanks for that. I'd seen some of the tool shortcuts on hover & sometimes added plugins also have shortcuts to. It would good is there was a definitive list somewhere.

GregValiant, 2024-04-21:
@gh13 The feature exists so can I go ahead and close this?

gh13, 2024-04-21:
@GregValiant I think the OP was asking for another feature to change the layer in preview so it's maybe them that deserves the response

GregValiant, 2024-04-21:
Sorry about that.  My bad.
@PeterVH72 how about you...is it OK to close this?

PeterVH72, 2024-04-21:
> @gh13 The feature exists so can I go ahead and close this?

Ok, go a head and close it ;)

GregValiant, 2024-04-21:
Done.

closed 2024-04-21 by GregValiant