Ultimaker/Cura #18922: Slizing failed

RoLuWa, 2024-04-20:
### Cura Version

5.6.0

### Operating System

Windows 10

### Printer

Anycubic Kobra Neo

### Describe model location

Direct Import, 0° Rotation or 90° Rotation CW

### Describe your model

distribution box cover

### Add your .zip here ⬇️

[Multisensorgehause_Deckel_0h-00m,_PLA+_Marc_0.2.3mf.zip](https://github.com/Ultimaker/Cura/files/15048575/Multisensorgehause_Deckel_0h-00m._PLA%2B_Marc_0.2.3mf.zip)

GregValiant, 2024-04-20:
Thanks for the report.
I didn't have any trouble slicing it and the model appears to be error free.

The Cura team will take a look.

RoLuWa, 2024-04-20:
Hi Greg,I have download the version 5.7.0 and the slicing is now also ok. Thanks for your supportRegards RolandVon meinem iPhone gesendet

GregValiant, 2024-04-20:
I'll go ahead and close this then.  There are a lot of fixes in 5.7.0 that resolve much of the slicing problems that plagued 5.4 to 5.6.

closed 2024-04-20 by GregValiant