Ultimaker/Cura #18924: Combining "Gradual Infill Steps" and "Infill Overlap Percentage" produces

daniels220, 2024-04-20:
### Cura Version

5.6.0

### Operating System

macOS Sonoma

### Printer

CR-10

### Reproduction steps

1. Start with default settings (which include an Infill Overlap Percentage), any reasonable model with some interior volume that will get infill.
2. Enable "Gradual Infill Steps", and values.

### Actual results

Observe little "dots" of infill appearing in the lower-density layers, where the full-density infill would/will overlap the walls.

### Expected results

Wall-overlap segments are only generated where actual infill appears.

### Add your .zip and screenshots here ⬇️

[curabug.3mf.zip](https://github.com/Ultimaker/Cura/files/15049557/curabug.3mf.zip)
[image]

GregValiant, 2024-04-20:
Thanks for the report.
I can duplicate this and I've seen it before.  It sort of makes the setting useless when the nozzle is spitting little dots everywhere.
[image]

[18924.zip](https://github.com/Ultimaker/Cura/files/15049774/18924.zip)

HellAholic, 2024-04-24:
Hey @daniels220,
Thanks for the report. Issue is resolved in 5.7.1 available to download now.
[image]

@GregValiant I also checked your model and no more ant trails. Magic performed by PR from Remco :)
[image]

Could you double check if it is the case for you as well?

GregValiant, 2024-04-24:
@HellAholic  Looks like it's fixed.  Congrats to Remco on the quick service.
(It does make me a little suspicious about where the bug came from.  Could it have been an "unexpected consequence" of something somebody did?)

@daniels220 this looks to be resolved in the new release.  I don't see the problem with 5.7.1.
Can we close this?

fieldOfView, 2024-04-24:
For reference:
https://github.com/Ultimaker/CuraEngine/pull/2062

daniels220, 2024-04-24:
Yep, fixed for me as well.

closed 2024-04-24 by daniels220