Ultimaker/Cura #18925: The little towers on top of this bridge prevent slicing

fplanque, 2024-04-20:
### Cura Version

5.7.0

### Operating System

MacOS arm64

### Printer

Creality Ender-5

### Name abnormal settings

I have sliced many other models with the same settings

### Describe model location

No, same issue in any orientation and position

### Describe your model

It's a bridge. It slices well if there is a tower on the top on only one side. As soon as there is a tower on each side, it can't slice any more. Slicing works in Orca.

### Add your .zip here ⬇️

[cura can't slice.zip](https://github.com/Ultimaker/Cura/files/15049659/cura.can.t.slice.zip)

GregValiant, 2024-04-20:
Thanks for the report.
I didn't have any trouble slicing it in Windows 10.  The model doesn't appear to have any errors.

Someone from the Cura team will take a look.

jellespijker, 2024-04-23:
Thank you for reporting this issue, when I try to slice it in our upcoming release candidate it seems to slice as expected, closing this as fixed.
[image]

closed 2024-04-23 by jellespijker