Ultimaker/Cura #18926: Slicing Failed (Cura 5.7)

csm10495, 2024-04-21:
### Cura Version

5.7.0

### Operating System

Windows 10

### Printer

Elegoo Neptune 3 Plus

### Describe your model

Socket adapters

### Add your .zip here ⬇️

[EN3P_Quarter_Inch_Drive_Ring.zip](https://github.com/Ultimaker/Cura/files/15050733/EN3P_Quarter_Inch_Drive_Ring.zip)
[image]

csm10495, 2024-04-21:
[EN3P_Quarter_Inch_Drive_Ring_works.zip](https://github.com/Ultimaker/Cura/files/15050746/EN3P_Quarter_Inch_Drive_Ring_works.zip)

This version works after I spread the parts out/messed with the settings. 

As for the error case, after opening the file and getting the failure: 

Bottom of cura.log says:

```
2024-04-20 22:33:01,194 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2024-04-20 22:33:01.195] [info] Slice of mesh took 0.003 seconds
2024-04-20 22:33:01,198 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2024-04-20 22:33:01.198] [info] Make polygons took 0.004 seconds
2024-04-20 22:33:01,206 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2024-04-20 22:33:01.206] [info] Progress: slice accomplished in 0.110s
2024-04-20 22:33:01,206 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2024-04-20 22:33:01.206] [info] Starting layerparts...
2024-04-20 22:33:01,667 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2024-04-20 22:33:01.667] [info] Progress: layerparts accomplished in 0.460s
2024-04-20 22:33:01,667 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2024-04-20 22:33:01.667] [info] Starting inset+skin...
2024-04-20 22:33:02,639 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2024-04-20 22:33:02.639] [info] Progress: inset+skin accomplished in 0.973s
2024-04-20 22:33:02,640 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2024-04-20 22:33:02.639] [info] Starting support...
2024-04-20 22:33:03,332 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2024-04-20 22:33:03.332] [info] Progress: support accomplished in 0.693s
2024-04-20 22:33:03,333 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2024-04-20 22:33:03.332] [info] Starting export...
2024-04-20 22:33:03,433 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2024-04-20 22:33:03.433] [error] The file %s could not be opened for writing.
2024-04-20 22:33:04,467 - ERROR - [MainThread] CuraEngineBackend.CuraEngineBackend._onBackendQuit [1038]: Backend exited abnormally with return code 3221226505!
2024-04-20 22:33:04,469 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [421]: Attempting to kill the engine process
2024-04-20 22:33:04,469 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [427]: Killing engine process
2024-04-20 22:33:04,469 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [430]: Engine process is killed. Received return code 3221226505
2024-04-20 22:33:04,470 - DEBUG - [MainThread] UM.Backend.Backend._createSocket [294]: Previous socket existed. Closing that first.
2024-04-20 22:33:04,479 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [232]: Socket state changed to Listening
2024-04-20 22:33:04,482 - INFO - [MainThread] UM.Backend.Backend.startEngine [96]: Started engine process: C:\Program Files\UltiMaker Cura 5.7.0\CuraEngine.exe
2024-04-20 22:33:04,482 - DEBUG - [MainThread] UM.Backend.Backend._backendLog [140]: [Backend] Calling engine with: ['C:\\Program Files\\UltiMaker Cura 5.7.0\\CuraEngine.exe', 'connect', '127.0.0.1:49674', '']
2024-04-20 22:33:04,510 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2024-04-20 22:33:04.509] [info] Sentry url: https://734f9ec9024f73e53701d59c3ffddfe3@o323038.ingest.sentry.io/4506257745510401
2024-04-20 22:33:04,510 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2024-04-20 22:33:04.509] [info] Sentry config path: C:/Users/*******/AppData/Roaming/cura/.sentry-native
2024-04-20 22:33:04,510 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2024-04-20 22:33:04.509] [info] Starting sentry
2024-04-20 22:33:04,519 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2024-04-20 22:33:04.518] [info] Connecting to 127.0.0.1:49674
2024-04-20 22:33:04,520 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [236]: Socket state changed to Connected
2024-04-20 22:33:04,520 - DEBUG - [MainThread] UM.Backend.Backend._onSocketStateChanged [225]: Backend connected on port 49674
2024-04-20 22:33:04,620 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2024-04-20 22:33:04.620] [info] Connected to 127.0.0.1:49674
```

darealfive, 2024-04-22:
There is a error in your log-files I also experience in my Cura 5.7 version on MAC-OS Sonoma 14.4.1 but I refuse to create a issue because github is full of the same issues.
The error we share is:

> The file %s could not be opened for writing.

jellespijker, 2024-04-23:
Thank you for reporting when I try to slice it in the build of our upcoming release, it slices correctly see zip file, closing this issue.

[EN3P_Quarter_Inch_Drive_Ring.zip](https://github.com/Ultimaker/Cura/files/15071541/EN3P_Quarter_Inch_Drive_Ring.zip)

closed 2024-04-23 by jellespijker